u-root/u-root #1483: [Question] By any chance, does u-root has any tool to inspect a compressed initramfs ?

10000TB, 2019-12-12:
Say for `someinitramfs.cpio.gz`, does uroot has a way to inspect what is inside ?

or better, some `diff` capability between two initramfs.cpio.gz ? 

say some process produced two initramfs from two builds, while new one has grown larger, and one need to do size impact analysis and find out what packages/libraries/? has attributed into the growth

hugelgupf, 2019-12-12:
you don't have to use u-root for this, you could for example:

gunzip initramfs.cpio.gz
cpio -ivt < initramfs.cpio

now, both of those tools should also be available in u-root itself:

gzip -d initramfs.cpio.gz using cmds/core/gzip
cpio t < initramfs.cpio using cmds/core/cpio

hugelgupf, 2019-12-12:
for the packages and libraries, it comes down to the busybox binary itself.

i tend to do one of two things

- compare binaries on their own, e.g. cmds/core/cpio before and after a change, (but this fails to capture the shared library savings of the busybox, of course)

- build two different initramfs, extract the bbin/bb binary, and analyze the differences in binary output

- use u-root makebb tool in `tools/makebb` to make busybox binaries without the initramfs. just builds the busybox binary itself to analyze and play with.

however, both u-root and makebb tools will produce stripped Go binaries. perhaps for debugging and analysis we should have an option to build them without the stripping of symbols and stuff?

hugelgupf, 2019-12-12:
for analyzing the binaries themselves, https://github.com/google/bloaty is a pretty good tool to start with.

hugelgupf, 2019-12-12:
I also just found this: https://github.com/jondot/goweight never used it, but looks interesting.

10000TB, 2019-12-12:
I could double check the ones you shared when I have access to my work environment, but I did have tried similar cmds to do decompression - it was also my intuition to approach the problem

but somehow I think I still saw a lot of binary-ish unreadable things after decompression... and lost hope from there.. and thought there might be some magic applied to the contents as well...but I am not entirely certain

hugelgupf, 2019-12-12:
Are you sure they were gz? The internal work ones tend to be xz (well, used to be anyway, now just uncompressed, and compressed by kernel).

dotwaffle, 2019-12-12:
I seem to have `lsinitramfs` installed on my machine... I don't know if that's installed by default or not. Running `lsinitramfs -l $file` should give you what you want.

10000TB, 2019-12-12:
I have `lsinitramfs` as well, 

I can see readable contents now.  Adding a endorsement to these steps

> you don't have to use u-root for this, you could for example:
> 
> gunzip initramfs.cpio.gz
> cpio -ivt < initramfs.cpio
> 
> now, both of those tools should also be available in u-root itself:
> 
> gzip -d initramfs.cpio.gz using cmds/core/gzip
> cpio t < initramfs.cpio using cmds/core/cpio

it is likely that the files are corrupted, or I was trying to cpio list compressed file (while by its name it is not compressed, like without `gz`)

closed 2019-12-12 by 10000TB