u-root/u-root #1499: Header files not copied to /go/pkg/include in initramfs

samvrlewis, 2019-12-30:
Hi,

I'm trying to get up and running with u-root but after building an initramfs with `u-root` (without any arguments) and then launching with qemu (with `qemu-system-x86_64 -nographic -kernel ~/Downloads/vmlinuz -initrd /tmp/initramfs.linux_amd64.cpio  -append "earlyprintk=ttyS0,115200,keep console=ttyS0" -m 256`) I get errors such as:

```
2019/12/30 08:30:29 Couldn't compile "init": error building go package in "/src/github.com/u-root/u-root/cmds/core/init": # internal/cpu
/go/src/internal/cpu/cpu_x86.s:8: #include: open /go/pkg/include/textflag.h: no such file or directory
```

Looking at the files within the initramfs, it looks as though the error message is correct.. there are no files within `/go/pkg/include`:

```
cpio -itv < initramfs.linux_amd64.cpio | grep go/pkg/include
drwxr-xr-x   0 root     root            0 Jan  1  1970 go/pkg/include
```

If I manually extract the files from the initramfs, `cp go/src/runtime/*.h go/pkg/include`, then repackage the initramfs, it seems to fix the problem but I'm guessing I shouldn't have to do this. Am I doing something wrong?

I'm using Ubuntu 18.04, will include the output of `go env` below:

```
GO111MODULE=""
GOARCH="amd64"
GOBIN=""
GOCACHE="/home/sam/.cache/go-build"
GOENV="/home/sam/.config/go/env"
GOEXE=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GONOPROXY=""
GONOSUMDB=""
GOOS="linux"
GOPATH="/home/sam/go"
GOPRIVATE=""
GOPROXY="https://proxy.golang.org,direct"
GOROOT="/usr/lib/go-1.13"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/usr/lib/go-1.13/pkg/tool/linux_amd64"
GCCGO="gccgo"
AR="ar"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD=""
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build287903227=/tmp/go-build -gno-record-gcc-switches"
```

rminnich, 2019-12-30:
I believe textflag.h always came in anyway or something. 
cpio from a run i did just now
-rw-rw-r--  0 root   root     1023 Jan 1 1970 go/pkg/include/asm_ppc64x.h
-rw-rw-r--  0 root   root     2418 Jan 1 1970 go/pkg/include/funcdata.h
-rw-rw-r--  0 root   root     1375 Jan 1 1970 go/pkg/include/textflag.h
rminnich@xcpu:~/gopath/src/github.com/u-root/webboot$ sudo `which testramfs` -i /tmp/initramfs.linux_amd64.cpio 
2019/12/30 17:15:19 Welcome to u-root!
               _
  _  _   _ __ ___  ___ | |_
 | | | |____| '__/ _ \ / _ \| __|
 | |_| |____| | | (_) | (_) | |_
  \__,_|  |_| \___/ \___/ \__|
(unreachable)/home/rminnich/gopath/src/github.com/u-root/webboot#
works fine
49109d71ef11429c3b2b6f0d0c9b10dfecc8ffc5
go version go1.12 linux/amd64

rminnich, 2019-12-30:
is GO11MODULE breaking it somehow?
are you building in GOPATH?

rminnich, 2019-12-30:
note that go modules broke source mode.

rminnich, 2019-12-30:
rminnich@xcpu:~/gopath/src/github.com/u-root/u-root$ cat /etc/issue
Ubuntu 18.04.3 LTS \n \l

rminnich@xcpu:~/gopath/src/github.com/u-root/u-root$ which go
/home/rminnich/go/bin/go

rminnich@xcpu:~/gopath/src/github.com/u-root/u-root$ /usr/bin/go version
go version go1.10.4 linux/amd64

so maybe build your own is the real fix.

rminnich, 2019-12-30:
this can also happen when one go version is used with a different go version's source tree IIRC?

samvrlewis, 2019-12-30:
> so maybe build your own is the real fix.

Hmm, it does work if I use a go root/binary that I built myself. 

If I run with either go-1.10 from the main Ubuntu repository or with go-1.12 from the ppa:longsleep/golang-backports repository nothing seems to get added to go/pkg/include.

Not sure what's going on but will close, looks like this is some sort of local configuration issue?

Thanks for the help.

hugelgupf, 2019-12-31:
Last time we had something like this, it had to do with how Ubuntu or Fedora configure their Go packages. We should get a VM and check it out, and perhaps have a Ubuntu VM CI build...

PR #1505 (2020-01-15): pkg/uroot: Fix adding files from a symlinked dir
This fixes #1499 where I was not able to build a working initramfs using an Ubuntu PPA provided golang version but was able to build a working initramfs using a golang version I built myself. The issue was that the PPA provided golang has a bunch of symlinks that the `addFile` function didn't appear to follow correctly.

```
$ ls /usr/lib/go-1.12/pkg/ -la
total 28
drwxr-xr-x  7 root root 4096 Aug 15 06:32 .
drwxr-xr-x  4 root root 4096 Jan  9 15:49 ..
lrwxrwxrwx  1 root root   34 Aug 15 06:32 include -> ../../../share/go-1.12/pkg/include
```
In  [... cut]
commit 81888348 (2020-01-15): pkg/uroot: Fix adding files from a symlinked dir

The addFile implementation should be able to handle directories that are
symlinked. This fixes the function to allow both symlinked files and
directories.

Fixes #1499.
closed 2020-01-15 by hugelgupf