u-root/u-root #1545: CircleCI: Integration of software tpm

ChriMarMe, 2020-02-05:
Hi everybody,

in behalf of @zaolin I plan to integrate swtpm into the CI so we can test tpm related code here as well.
What do you think?

I'm still beginner in working with CI and especially CircleCI, so I may need some guidance.

Greetings

hugelgupf, 2020-02-06:
That sounds like an awesome idea to me. swtpm integrates with QEMU, right?

All of our CircleCI QEMU testing uses the docker image built by [this Dockerfile](https://github.com/u-root/u-root/blob/master/.circleci/images/test-image-amd64/Dockerfile). If you want to modify QEMU to have swtpm for the tests, you should first modify that docker image.

ChriMarMe, 2020-02-06:
https://github.com/ChriMarMe/u-root/tree/CIswtpm

Thats the first step. 
Should I open a PR for communication and close the issue or how do we proceed?

hugelgupf, 2020-02-06:
We can keep using this. Have you looked at how our integration tests are written?

ChriMarMe, 2020-02-07:
Not yet, but I'll take a deep dive head first into it, now that I know where to look :>

ChriMarMe, 2020-02-07:
After I looked at it, I'm a little confused. 

My main concern right now is to set up the swtpm in a way that it either run outside qemu but can be accessed from inside qemu or runs inside qemu (so kernel & initramfs needs the necessary packages preinstalled)

Both seems not as easy as it is said.

ChriMarMe, 2020-03-03:
Alright, after three weeks of studying for exams I'm back at work and digging into it again.

ChriMarMe, 2020-03-04:
We may get #1493 merged before writing integration tests for that. Still need to figure out if it is possible to hand over two tpm devices to qemu at the same time. Also this is not my top prio.