u-root/u-root #1563: uio lazyopener segv

rminnich, 2020-02-20:
I think I might have had this some time ago, never resolved, here it is again:

~/> kexec -l arch/x86//boot/bzImage 
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x553666]

goroutine 1 [running]:
github.com/u-root/u-root/pkg/uio.(*LazyOpenerAt).ReadAt(0x0, 0xc00018c000, 0x8000, 0x8000, 0x0, 0xc0c501, 0xc00018c000, 0x8000)
	/usr/local/google/home/rminnich/go/src/github.com/u-root/u-root/pkg/uio/lazy.go:89 +0x26
io.(*SectionReader).Read(0xc0000fa930, 0xc00018c000, 0x8000, 0x8000, 0x0, 0x0, 0xc000000180)
	/usr/lib/google-golang/src/io/io.go:478 +0x7c
io.copyBuffer(0xd69060, 0xc00000e188, 0xd68bc0, 0xc0000fa930, 0xc00018c000, 0x8000, 0x8000, 0x4, 0xc000037700, 0x15)
	/usr/lib/google-golang/src/io/io.go:405 +0x122
io.Copy(0xd69060, 0xc00000e188, 0xd68bc0, 0xc0000fa930, 0xc00000e188, 0x0, 0x0)
	/usr/lib/google-golang/src/io/io.go:364 +0x5a
github.com/u-root/u-root/pkg/boot.copyToFile(0xd68bc0, 0xc0000fa930, 0x0, 0x0, 0x0)
	/usr/local/google/home/rminnich/go/src/github.com/u-root/u-root/pkg/boot/linux.go:49 +0xbc
github.com/u-root/u-root/pkg/boot.(*LinuxImage).Load(0xc000086800, 0xc000086800, 0x0, 0x0)
	/usr/local/google/home/rminnich/go/src/github.com/u-root/u-root/pkg/boot/linux.go:99 +0x369
github.com/u-root/u-root/cmds/core/kexec/%2ebb.Main()
	/usr/local/google/home/rminnich/go/src/github.com/u-root/u-root/cmds/core/kexec/.bb/kexec_linux.go:107 +0x317
github.com/u-root/u-root/pkg/bb/bbmain.Run(0x7ffdb7ceaebe, 0x5, 0x7ffdb7ceaebe, 0x5)
	/usr/local/google/home/rminnich/go/src/github.com/u-root/u-root/pkg/bb/bbmain/register.go:62 +0x8e
main.run()
	/usr/local/google/home/rminnich/go/src/github.com/u-root/u-root/bb/main.go:115 +0x6f
main.main()
	/usr/local/google/home/rminnich/go/src/github.com/u-root/u-root/bb/main.go:123 +0x7b
Exception: kexec exited with 2
[tty], line 1: kexec -l arch/x86//boot/bzImage 
~/> ls -l arch/x86//boot/bzImage 
-rw-r----- 0 0 7232848 Jan  1 00:00 bzImage
~/>

rminnich, 2020-02-20:
It is actually creating two files, on zero length, one the full length, in /tmp

rminnich, 2020-02-20:
go 1.13

rminnich, 2020-02-20:
ah ok. It is because the initrd is ""

hugelgupf, 2020-02-21:
the lazy opener should still not segv... it should handle the error gracefully and be able to return it through ReadAt