u-root/u-root #1610: pkg/ulog: KernelLog initialization

jensdrenhaus, 2020-03-24:
I don't understand /pkg/ulog/klog_linux.go line 39: 
```
// Reinit reopens the /dev/kmsg file.
func (k *KLog) Reinit() {
	f, err := os.OpenFile("/dev/kmsg", os.O_RDWR, 0)
	if err != nil {
		KernelLog.File = f
	}
}
```
If os.OpenFile returns a non-nil Error, `f` allways will be nil. With this implementation `ulog.KernelLog.Print("HALLO WORLD")` has no effect in my programs.

I changed it for testing like so:
```
var x = 1

// Reinit reopens the /dev/kmsg file.
func (k *KLog) Reinit() {
	log.Printf("KLOG CALL %d", x)
	x++
	f, err := os.OpenFile("/dev/kmsg", os.O_RDWR, 0)
	if err != nil {
		log.Printf("KLOG ERROR %v %v", f, err)
		KernelLog.File = f
	}
	KernelLog.File = f
}
```
Now, `HELLO WORLD` is written to /dev/kmsg.
Further I can see that Reinit() is executed multiple times. (three times with x=1, one time with x=2 in my case)
I thought package initialization is done only once even if the package is imported many times.

Can you help me what was the initial idea with the assignment of KernelLog.File, please.

rminnich, 2020-03-24:
I don't know what's up there but that code looks wrong to me.
Suspect it was supposed to be == nil and somebody's fingers autottype != nil

hugelgupf, 2020-03-25:
Ron is pretty much right.

This was meant for kernels that don't automount devtmpfs, so at init() time /dev/kmsg doesn't exist yet. When you do finally mount devtmpfs in pkg/libinit, it calls this Reinit.

Seems it never worked though 🤡

Yeah, it's supposed to be `==`.

jensdrenhaus, 2020-04-01:
okay -> #1618

PR #1618 (2020-04-01): pkg/ulog: fix KernelLog Reinit()
fixes #1610
closed 2020-04-01 by rminnich