u-root/u-root #1627: Fix CirrusCI or throw it awy

ChriMarMe, 2020-04-06:
Hey everybody,

what's up with CirrusCI?

Every PR generates errors in setting up the env in CirrusCI. 
Any suggestions? 

Greetings
Chris

rjoleary, 2020-04-06:
The error I keep seeing is:

```
pkg install -y go dep git
Updating FreeBSD repository catalogue...
Fetching meta.txz: . done
pkg: repository meta /var/db/pkg/FreeBSD.meta has wrong version 2
repository FreeBSD has no meta file, using default settings
Fetching packagesite.txz: .......... done
pkg: repository meta /var/db/pkg/FreeBSD.meta has wrong version 2
pkg: Repository FreeBSD load error: meta cannot be loaded No error: 0
Unable to open created repository FreeBSD
Unable to update repository FreeBSD
Error updating repositories!

Exit status: 3
```

For now, I'll make CirrusCI non-required, so you can still push your changes even if CirrusCI fails.

ChriMarMe, 2020-04-07:
Thank you good sir :)

closed 2020-04-07 by ChriMarMe