u-root/u-root #1645: Strategy for keeping Elvish updated to upstream?

xiaq, 2020-04-21:
Hi, I am the main developer of Elvish here, and I noticed that the Elvish version in u-root is now pretty old; there has been a lot of changes recently.

I am interested in hearing about your strategy of keeping Elvish updated. So far there seems to be two main divergences:

* u-root's fork does not have the daemon.

  This can be avoided by some refactoring on Elvish's end; in the internal API the daemon is already [optional](https://github.com/elves/elvish/blob/6938c8d11701555548cf19208087361f9397ad4a/pkg/program/shell/script.go#L17). It requires some more work to be able to a version of Elvish without the daemon from the same codebase though, but it should be doable.

* u-root's fork does not use `reflect.Call`.

  This is much harder to conciliate. I expect that you may need to continue redoing this change when merging Elvish.

I can see some other smaller [divergences](https://github.com/u-root/u-root/commits/master/cmds/core/elvish) - these that can be upstreamed instead of committed in u-root's fork (and I encourage you to do so!).

xiaq, 2020-04-21:
It seems that there is ongoing work in the Go team to unbreak dead code elimination when reflect.Value.Call is used: golang.org/cl/228792

xiaq, 2020-04-21:
I am also interested in the rationale of removing the daemon - is it because (1) you don't want to run a daemon process, or (2) you want to trim down the binary by removing the boltdb code?

xiaq, 2020-04-21:
I found this comment from one of the u-root contributors about the daemon: https://github.com/elves/elvish/issues/425#issuecomment-314319968. I explained that the use of Unix socket is OK, but did not hear back.

hugelgupf, 2020-04-21:
Re removing the daemon - it's both. We care a *lot* about binary size, so we trim where possible. elvish at the point we forked had all the features we needed, so it hasn't been in our interest to update yet - what does upstream have now that you think would be good to have?

xiaq, 2020-04-21:
You can find changes introduced by the recent release notes in https://elv.sh/blog/.

In general, it can be quite confusing for users if the Elvish fork in u-root is very different from the actual Elvish project. Elvish is an evolving project, and an old version is essentially a different program; this is very different from say Go, which has been more or less stable.

hugelgupf, 2020-04-21:
What if we called it felvish? fake elvish? I think a couple of replacements for elvish have been in progress for a while, but we've never put much of a priority on it, because our trimmed down elvish has been good enough for embedded systems.

I think we'd be happy to upstream things, but I'm guessing you aren't planning on putting an emphasis on binary size?

xiaq, 2020-04-24:
Rebranding definitely removes my concern about confusing users. Making it clear that the Elvish shipped with u-root is an old version and bugs should be reported to u-root would also works without rebranding. Up to you.

However, I am still interested in helping you sync to a newer Elvish version, and this does not need to be a one-way relationship. I don't see u-root shipping a vanilla Elvish at the moment, but I am open to make changes to minimize the changes you need to make downstream.

xiaq, 2020-04-24:
As mentioned in my original post, some refactoring can be done to make the daemon truly optional. If the integration point is in the main package, u-root does not need to do "surgical" changes to the code to remove the daemon, but rather, just a different main package.

volfyd, 2020-08-06:
I have definitely run into this problem, because I'm trying to get this line to work:

elvish -c 'for i [i915 atkbd i8042] { /usr/bin/modprobe -v $i }'

and I have really no way of figuring it out, because the version of 'elvish' included in u-root is not documented anywhere Google takes me. I lost like 45 minutes on this, and my solution is to just include busybox in my image so I can write a simple for loop.

I hope you guys figure this out, because changing the name doesn't magically cause documentation to appear at the top of Google, so it's just one more thing to hunt down and for you to maintain.

hugelgupf, 2020-08-06:
Hey, that's a good point. Just to leave this here - we do not really intend to support much shell scripting in u-root. It should be easy enough to recompile that even a small shell script ends up being a Go program instead. Whatever shell is in our ultimate end state should (in my view) not support conditionals or loops.

Some time ago, we actually had kind of a go repl. That has kind of died with our source mode becoming a second class citizen. So maybe... it's worth reconsidering our stance on the above, actually.

Either way, there or not, hear ya on better docs.

krader1961, 2021-05-30:
FYI, a u-root user recently opened an Elvish project [issue](https://github.com/elves/elvish/issues/1324) asking why tab completion did not work as they expected. It does work as they expect in the current Elvish release. I believe the Elvish project owner is willing to make any reasonable change that would allow this project to use the `elvish` binary without having to embed its source into this project and manually patch it. In light of this [comment](https://github.com/u-root/u-root/issues/1645#issuecomment-669669465) does it make sense to do the work necessary in both projects to make it easy for u-root to embed a current version of Elvish?

rjoleary, 2021-06-07:
The size difference looks pretty substation (this is core + elvish):

```
With go version 1.13:
16571556 new_elvish.cpio
 4398452 new_elvish.cpio.xz
15309988 old_elvish.cpio
 4052772 old_elvish.cpio.xz

With go version 1.16:
13679780 new_elvish.cpio
 3519348 new_elvish.cpio.xz
11701412 old_elvish.cpio
 3059740 old_elvish.cpio.xz
```

At the same time, the size optimizations in the newer Go compiler look quite good, so maybe we can take the hit?

krader1961, 2021-06-07:
@rjoleary: I believe the biggest reason for the size difference is the support for the Elvish interactive history daemon which your hand optimized version elides. It might be possible for Elvish to provide that optimization via a build option.

On the other hand, I have been lobbying (see https://github.com/elves/elvish/issues/126#issuecomment-615543154 and https://github.com/elves/elvish/issues/1222) for changes that would eliminate the daemon by replacing the `bbolt` database with `mysql` or a pure Elvish mechanism; thus making the daemon unnecessary. It's unclear how big the difference in size would be with either of those changes but it would eliminate all the overhead related to the Elvish daemon and result in a smaller Elvish binary and run-time footprint. It may be better long term to encourage that approach rather than the short term solution of eliding the daemon from a u-root build of the Elvish binary.

rjoleary, 2021-06-07:
#2038 seems to work. Is it possible to remove these warnings?

```
Cannot connect to daemon: daemon unreachable after waiting for 1s
Daemon-related functions will likely not work.
```

krader1961, 2021-06-07:
> Is it possible to remove these warnings?

Yes, but that will require either 

1) a build-option, or equivalent, that u-root can use, to eliminate (or stub) Elvish daemon behavior, or

1) implementing either of the changes, switching to pure Elvish or non-cgo Mysql, that would allow the Elvish daemon to be eliminated.

krader1961, 2021-06-07:
> Is it possible to remove these warnings?

It might also be possible to elide those warnings if there is a reliable way for Elvish to determine it is running in a u-root instance. I can't guarantee the project owner, @xiaq, will agree to that approach but it warrants discussion.

rminnich, 2021-06-07:
the daemon has to be removable at build time and removing it must not cause errors. If that is possible then I'm fine with it.

rminnich, 2021-06-07:
further, any sort of error messages related to the missing deamon are unacceptable. We'd need to fix this.

If the deamon is only for history, couldn't we make history an interface and implemented it different ways?

rminnich, 2021-06-07:
if engineers at the companies we work with see:
Cannot connect to daemon: daemon unreachable after waiting for 1s
Daemon-related functions will likely not work.

we'll get lots of complaints ...

krader1961, 2021-06-08:
@rminnich: I think we're all on the same page. The need for the interactive history daemon should be eliminated or it should be possible to build elvish without support for the daemon. So that running `elvish` in an interactive context works without artifacts like unexpected warning messages due to the daemon not being supported or available.

rjoleary, 2021-06-10:
If there was a copy of the main package called `elvish-lite` where `deamon.Program` is removed from this line:

https://github.com/elves/elvish/blob/master/cmd/elvish/main.go#L19,

would that mean the Daemon would never start and never be called?

krader1961, 2021-06-10:
@rjoleary, That would keep the daemon from being started but would result in unwanted error messages. It would also leave the elvish binary with a lot of code (e.g., the bbolt DB and rpc support) that would be unused. What's really wanted is to create a variant of the daemon package that stubs the public client API and removes all the unneeded dependencies.

xiaq, 2021-06-10:
Before discussing more how to stub out the daemon, I want to raise awareness to bullet point 2 in my original comment - u-root avoids `reflect.Call` since it (used to?) break Go compiler's dead code elimination. Is that still the case now, i.e. does u-root still avoids any use of `reflect.Call` as a policy?

krader1961, 2021-06-11:
Like @xiaq I also wonder if eliminating uses of `reflect.Value.Call` in the Elvish source is still necessary as illustrated by this change: https://github.com/u-root/u-root/commit/a4c8954ddd8579fbfaa40b217c01ea1b6ea5d08c#diff-60fbc218051a48d4cf10972452c4eaa54e467262121a2d4751f31b34eff7c021. My work in progress change to eliminate the daemon does not eliminate those reflection calls. My change does eliminate the daemon. But it does not eliminate all If eliminating all uses of reflection are

krader1961, 2021-06-15:
If https://github.com/elves/elvish/pull/1337 is merged as is you'll be able to build an elvish binary with no daemon (and thus no RPC code or creation of a socket) by running `make get ELVISH_MAKE_BIN=./elvish ELVISH_MAKE_TAGS=elv_daemon_stub`. This reduces the size of the binary 15% on macOS.

TODO:  eliminating the unused bbolt DB code.

krader1961, 2021-06-16:
> TODO: eliminating the unused bbolt DB code.

I implemented that as well. I haven't pushed it for review because it shaves only 66K on macOS (64K on Linux x64) from the size of the elvish binary. That's merely a 0.9% decrease in size; which is hard to justify given that it adds 127 lines of code and requires running this command before building elvish to take advantage of the bbolt elimination:

```
go mod edit -replace go.etcd.io/bbolt ./bbolt_stub
```

A much more radical refactoring of the code would avoid the need for the go.mod replace directive but that is even harder to justify.

krader1961, 2021-06-20:
@rjoleary, @rminnich: The Elvish project owner was spurred by my proof of concept change to implement an approach they found more palatable; i.e., does not involve custom build tags. You can now generate an elvish binary with no interactive shared history daemon by executing `go build ./cmd/nodaemon/elvish` using the current master branch. The resulting elvish binary is 16% smaller (7.9M => 6.6M) on macOS and does not have any unwanted artifacts such as unexpected error messages due to the absence of the daemon.

rjoleary, 2021-08-03:
I am getting the following error:

```
--- FAIL: TestUrootCmdline (221.35s)
    --- FAIL: TestUrootCmdline/make_sure_dead_code_gets_eliminated (30.64s)
        uroot_test.go:320: Commandline: [-o /tmp/u-root-292472354 -build=bb -no-strip world github.com/u-root/u-root/pkg/uroot/test/foo -github.com/u-root/u-root/cmds/exp/builtin -github.com/u-root/u-root/cmds/exp/run]
        uroot_test.go:286: validator failed: Unused non-interface method has not been eliminated, dead code elimination is not working properly.
            The most likely reason is use of reflect.Value.Method or .MethodByName somewhere (could be a command or vendor dependency, apologies for not being more precise here).
            See https://golang.org/src/cmd/link/internal/ld/deadcode.go for explanation.
```

See https://github.com/u-root/u-root/pull/2084

krader1961, 2021-08-03:
@rjoleary, Elvish makes extensive use of the reflect package. There are two places that use `reflect.Value.Method` even when the daemon support code is not included. See *pkg/eval/vals/struct_map.go*. It might be possible to eliminate those two uses but unless doing so is straightforward it's unlikely to happen. Elvish also uses `reflect.Call` in other places that probably can't reasonably be replaced by a different implementation. Have you actually built elvish using `go build ./cmd/nodaemon` and inspected the resulting binary to see if it is acceptable without running your dead code elimination process?

rjoleary, 2021-08-03:
The size different is pretty significant:

**Before:**

```
~/go/src/github.com/u-root/u-root/cmds/core/elvish$ go build -ldflags='-s -w' . && cat elvish | xz | wc -c
1054000
```

**After:**

```
~/go/src/github.com/u-root/u-root/cmds/core/elvish$ go build -ldflags='-s -w' . && cat elvish | xz | wc -c
1895652
```

If the linker's dead code elimination isn't being run, it'd explain this.

What's weird is I've tried commenting out all the `Method` and `MethodByName` functions I could find with `git grep `\bMethod('` and `git grep '\bMethodByName('`, but the test still fails.

krader1961, 2021-08-03:
What is your CWD when you do that build? It would need to be the *./cmd/nodaemon/elvish* directory to get an elvish binary without support for the daemon. Also, the project has grown a lot since you last forked the code. Most notably by the inclusion of a lot more builtin functionality via the addition of new modules: https://elv.sh/ref/. That's going to mean more of the Go stdlib is included now compared to your forked version.

rjoleary, 2021-08-04:
When I run `CGO_ENABLED=0 go build -ldflags -c .`, I don't see anything using `Method` or `MethodByName`.

krader1961, 2021-08-05:
When I run `` on macOS I see this in the output:

```
reflect.(*rtype).Method calls runtime.duffzero
reflect.(*rtype).Method calls reflect.(*interfaceType).Method
reflect.(*rtype).Method calls runtime.duffcopy
reflect.(*rtype).Method calls runtime.duffcopy
reflect.(*rtype).Method calls reflect.(*rtype).exportedMethods
reflect.(*rtype).Method calls reflect.resolveNameOff
reflect.(*rtype).Method calls reflect.resolveTypeOff
reflect.(*rtype).Method calls runtime.makeslice
reflect.(*rtype).Method calls runtime.gcWriteBarrierDX
reflect.(*rtype).Method calls runtime.growslice
reflect.(*rtype).Method calls runtime.makeslice
reflect.(*rtype).Method calls runtime.gcWriteBarrierR9
reflect.(*rtype).Method calls runtime.growslice
reflect.(*rtype).Method calls reflect.(*rtype).IsVariadic
reflect.(*rtype).Method calls reflect.FuncOf
reflect.(*rtype).Method calls runtime.newobject
reflect.(*rtype).Method calls reflect.resolveTextOff
reflect.(*rtype).Method calls runtime.gcWriteBarrier
reflect.(*rtype).Method calls runtime.gcWriteBarrierR8
reflect.(*rtype).Method calls runtime.growslice
reflect.(*rtype).Method calls runtime.panicdottypeI
reflect.(*rtype).Method calls runtime.panicSlice3C
reflect.(*rtype).Method calls runtime.panicSlice3C
reflect.(*rtype).Method calls runtime.panicSlice3Alen
reflect.(*rtype).Method calls runtime.panicSlice3Alen
reflect.(*rtype).Method calls runtime.gopanic
reflect.(*rtype).MethodByName calls runtime.duffzero
reflect.(*rtype).MethodByName calls reflect.(*rtype).uncommon
reflect.(*rtype).MethodByName calls reflect.resolveNameOff
reflect.(*rtype).MethodByName calls runtime.duffzero
reflect.(*rtype).MethodByName calls runtime.memequal
reflect.(*rtype).MethodByName calls reflect.(*rtype).Method
reflect.(*rtype).MethodByName calls runtime.duffcopy
reflect.(*rtype).MethodByName calls runtime.duffcopy
reflect.(*rtype).MethodByName calls runtime.duffzero
reflect.(*rtype).MethodByName calls runtime.duffzero
reflect.(*rtype).MethodByName calls reflect.(*interfaceType).MethodByName
reflect.(*rtype).MethodByName calls runtime.duffcopy
reflect.(*rtype).MethodByName calls runtime.duffcopy
reflect.(*rtype).MethodByName calls runtime.duffcopy
reflect.(*rtype).MethodByName calls runtime.panicSlice3Alen
reflect.(*rtype).MethodByName calls runtime.morestack_noctxt
[many more like the above]
src.elv.sh/pkg/eval/vals.Kind calls reflect.Value.MethodByName
src.elv.sh/pkg/eval/vals.Kind calls reflect.Value.MethodByName
src.elv.sh/pkg/eval/vals.(*structMapIterator).Get calls reflect.Value.Method
```

krader1961, 2021-08-05:
@rjoleary, I just checked a Linux distro and see the same output I saw on macOS. I think we need more information about the commands you are running, the CWD, and the distro you are using. Elvish most definitely uses the `reflect.Value.Method` and `reflect.Value.MethodByName` functions -- even for the *cmd/nodaemon/elvish* build. There is no practical way to change that without breaking the public API; i.e., the Elvish language specification. So if eliminating those `reflect` package functions from the resulting binary is mandatory it seems likely you will have to stick with your custom, ancient, fork.

MDr164, 2021-11-16:
So as discussed with Ryan and Ron I'll prepare a nodaemon build that should replace the fork. After using xz compression the file increased by a substantial 372kb which for most machine won't make a big difference but especially devices with smaller flash chips it could be troublesome. It's only a matter of handling the reflect package now, so we could either make the check in the CI non-blocking or find a clever way around it which I don't yet see other than breaking the public API to some degree.

rjoleary, 2022-01-04:
We are using the updated elvish now: https://github.com/u-root/u-root/tree/main/cmds/core/elvish

closed 2022-01-04 by rjoleary

xiaq, 2022-01-04:
Thanks! I see that you are now taking on Elvish as a dependency rather than forking Elvish's code.

The latest release is now 0.17.0 - any chance that the dependency could be bumped?