u-root/u-root #1737: pkg/checker - LinkSpeed incorrect for extremely fast interfaces

mpictor, 2020-06-26:
Noticed while looking for ethtool-related code, haven't actually tested.

at https://github.com/u-root/u-root/blob/master/pkg/checker/interfaces.go#L43
```go
		if int(eth.Speed) < minSpeed {
			return fmt.Errorf("link speed %d < %d", eth.Speed, minSpeed)
		}
```
`eth.Speed` is a uint16 and should be used in conjunction with `eth.Speed_hi`; the ethtool package calculates this and its `EthtoolCmd.CmdGet()` returns it. However, that value is thrown away by this package.

----

Also, I noticed (as of ethtool 5.4) that `struct ethtool_cmd` is deprecated and `struct ethtool_link_settings` should be used instead. The ethtool package's `EthtoolCmd` struct corresponds to ethtool_cmd. I created an issue there noting this: https://github.com/safchain/ethtool/issues/33

mpictor, 2020-06-26:
@insomniacslk

insomniacslk, 2020-06-26:
thank you @mpictor ! I'll look at this in my upcoming restructuring of this framework

PR #1739 (2020-07-08): [pkg/checker] Fix retrieval of link speed
Fixes #1737

Tested locally:
```
$ go build && sudo ./fixmynetboot enx4865ee148371
 -> no remediation found for enx4865ee148371 link speed, skipping.
 -> no remediation found for enx4865ee148371 has link-local, stop on error requested. Exiting.
2020/06/29 01:04:50 no link local addresses for interface enx4865ee148371
```
commit 7a9efec9 (2020-07-08): [pkg/checker] Fix retrieval of link speed

Fixes #1737

Tested locally:
```
$ go build && sudo ./fixmynetboot enx4865ee148371
 -> no remediation found for enx4865ee148371 link speed, skipping.
 -> no remediation found for enx4865ee148371 has link-local, stop on error requested. Exiting.
2020/06/29 01:04:50 no link local addresses for interface enx4865ee148371
```
closed 2020-07-08 by probot-auto-merge[bot]