u-root/u-root #1740: add Measure() - method to boot.OSImage interface

jensdrenhaus, 2020-06-29:
Does it make sense to extend the Interface like the following?
```
// OSImage represents a bootable OS package.
type OSImage interface {
	fmt.Stringer

	// Label is a name or short description for this OSImage.
	//
	// Label is intended for boot menus.
	Label() string

       // Measure measures OS image into PCR 
       // 
       // Measure is intended for a measured boot environment.
       Measure() error

	// Load loads the OS image into kernel memory, ready for execution.
	//
	// After Load is called, call boot.Execute() to stop Linux and boot the
	// loaded OSImage.
	Load(verbose bool) error
}
```
I can do the implementation for `boot.LinuxImage` and `boot.MultibootImage` using `crypto.TryMeasureData`.

hugelgupf, 2020-06-29:
Yes it does. I've been thinking about something like this.

The only thing I'd add is `Measure(*tss.TPM)` for the unlikely case that there is more than one.

Also, is the PCR that pieces are measured into always the same?

JulienVdG, 2020-06-30:
This means that the Measure function has to be implemented for any OSImage type, is it what you want ?
Alternatively the Measure function could be in another interface... my 2 cents

hugelgupf, 2020-06-30:
I think it could be a separate, optional interface that any OSImage may implement, but doesnt have to, if that worries you.

hugelgupf, 2020-06-30:
In fact, this is where we could start intersecting with the securelaunch stuff that Oracle has contributed, and make the Collector interface work for us: https://github.com/u-root/u-root/blob/022ea728d31df46c0b91a317fa1901c713f907b8/pkg/securelaunch/measurement/collectors.go#L27