u-root/u-root #1799: Potential code quality issues found

chsatyap, 2020-08-13:
I forked this repo a while ago and ran a DeepSource analysis on it. DeepSource found a variety of different issues categorized based on their types and severity which you can view [here](https://deepsource.io/gh/chsatyap/u-root/).

Brief description -
Anti-Patterns: 120
Bug Risks: 1K
Performance issues: 6
Security Issues: 99

You can find a detailed description and fixes for some of them [here](https://deepsource.io/gh/chsatyap/u-root/issues/?category=recommended).

Please let me know what issues you'd like to focus on fixing and I'd be happy to take a look into it. Also, you can choose to hide certain types of issues too (if you wish to ignore them or you believe it is a false positive). I'd also be happy to send a patch with the DeepSource configuration file with the required fixes.

You can find the required configuration file for DeepSource [here](https://github.com/chsatyap/u-root/blob/master/.deepsource.toml).

rminnich, 2020-08-13:
I wonder if you could take each finding, produce a PR for it, and let's see about getting that change in?

rjoleary, 2020-09-17:
This looks like it can be very useful!
Do you know if there's any way to automatically create a PR or commit from this tool?

For example, the 800 unused code findings (https://deepsource.io/gh/chsatyap/u-root/issue/SCC-U1000/occurrences), it would take me the better part of a week to open each file, fine each line and delete. And, it seems like it would be easy for the computer to quickly generate a change from these findings. I'd still want to manually scan through, but this would definitely speed up the process.

I see there's a blog post about autofix (https://deepsource.io/blog/autofix-future-of-code-reviews/), but I couldn't find the button.

closed 2021-09-20 by rjoleary