u-root/u-root #1809: Tracking image sizes

rojer, 2020-08-18:
We need to be conscious of the size of the output images.
Image size impact of every change should be evaluated during review.
Things like #1798 do happen and go unnoticed while it would've probably raised a red flag in review if the size impact was more apparent (#1808 specifically addresses the DCE regression but a size bump should've raised a red flag as well).

I think every PR should come with a size diff breakdown for standard templates (minimal, core, all).
I am going to work on this.
Rough idea is:

1. Gather stats during CircleCI build, e.g. [this part](https://github.com/u-root/u-root/blob/60aeb0ab57dfac6e19f057de0b3e25793ede1616/.circleci/config.yml#L173-L187), and store as a build artifact.
2. Some sort of automation (new GitHub action?) to automatically comment on PRs with comparison to last master build stats, e.g. "all: +12345 bytes, core: no change, minimal: no change, ...".

cc @insomniacslk

rojer, 2020-08-20:
@hugelgupf @rminnich after #1810 goes in, the intention is to have CircleCI post stats to pull request after successful build as a comment.
for that, a GitHub token is required. do we have one already that can be used for this? or can one be created and added as a project-level variable?

rojer, 2020-09-14:
@hugelgupf thak you for merging #1810, we are collecting the stats now but to provide diffs and PR feedback we need DircleCI to carry a GH token with the necessary permissions. can you help setting one up and putting it into a variable?

hugelgupf, 2020-09-14:
Hey, we can set one up. What exactly does the posting?

rojer, 2020-09-14:
my current plan is to add a step to the CircleCI build when building a branch other than master, after collecting stats, to fetch latest master build stats via CircleCI API, diff them and post a comment to the PR via GH API. for that to work, CircleCI build environment should have access to a GH token with enough permissions to be able to comment on PRs.

rojer, 2020-09-27:
@hugelgupf what do you think of this? do you have other suggestions?

rjoleary, 2021-06-07:
Are you referring to this for posting a comment? https://support.circleci.com/hc/en-us/articles/360048170573-Auto-comment-on-GitHub-Pull-Requests

Aren't these GitHub tokens secret and not safe to put in the CI?