u-root/u-root #1815: Azure pipeline builds against external version of packages

rojer, 2020-08-26:
It looks like the Azure pipeline uses external imports when building u-root.go

Examples: [one](https://dev.azure.com/rminnich/u-root/_build/results?buildId=3913&view=logs&j=12f1170f-54f2-53f3-20dd-22fc7dff55f9&t=dc34f6f3-1287-512a-fdda-9b8da42a9877), [two](https://dev.azure.com/rminnich/u-root/_build/results?buildId=3895&view=logs&j=12f1170f-54f2-53f3-20dd-22fc7dff55f9&t=dc34f6f3-1287-512a-fdda-9b8da42a9877).

In both cases there was a change in `pkg` that `u-root.go` is using and all other builds except Azure are passing. The failure indicate that new u-root.go is being built against old `pkg`.

cc @rminnich

rminnich, 2020-08-26:
I dropped azure on another project. It's no longer worth the effort IMHO even though I liked testing builds against osx. reading the azure docs it seems they had to do a lot to make it work with github user and permission models.

flokli, 2021-03-04:
Is azure pipelines still configured? I do see the file in the repo, but I don't see any checks from it attached to master builds…

rjoleary, 2021-05-27:
See https://github.com/u-root/u-root/issues/2021 for discussion on removing Azure pipelines CI.

rjoleary, 2021-06-07:
Is #2034 goes through, this can be closed.

closed 2021-06-07 by rjoleary