u-root/u-root #1835: failed to boot into CentOS 8 with 'boot' command due to incorrect kernel cmdline

gabhuang, 2020-09-17:
CentOS 8 uses Boot Loader Specification config with $ variables.
the kernel options in loader/entries/*.conf:

_options $kernelopts $tuned_params_

current bls parser don't know how to deal with $ variables which defined in grub.cfg or grubenv.

_# GRUB Environment Block
saved_entry=1
kernelopts=root=UUID=1232991f-d630-4071-9d01-135cadd64ba3 ro  crashkernel=auto_ 

dmesg when booting:
[    0.000000] Kernel command line: $kernelopts console=ttyAMA0,115200
Welcome to [0;34mCentOS Linux 8 (Core) dracut-049-27.git20190906.el8 (Initramfs)[0m!

[   11.873578] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000010
then kernel panic.

config files here:
[centos8_config_files.zip](https://github.com/u-root/u-root/files/5237134/centos8_config_files.zip)

hugelgupf, 2020-09-17:
Hey,

yeah, we are acutely aware of this problem, and I'm sorry for the deficiency. To be clear, Fedora 28 specifically chose to abuse the BootLoaderSpec in a non-spec-compliant way, as detailed [here](https://fedoraproject.org/wiki/Changes/BootLoaderSpecByDefault) ([Lennart's comment on that here :|](https://github.com/systemd/systemd/issues/15029#issuecomment-607691208)). As far as I know, it's going away with Fedora 33 -- but that doesn't help us here, since we'll be stuck with compatibility for a while. CentOS 8 + RHEL 8 both picked up this boot loader design.

The "right" way to do this would be to implement a full blown GRUB parser that supports the `bls` plugin -- the GRUB parser defines those grubenv variables and then plugs them into the BLS as you showed.

Unfortunately a fully-featured Turing-complete GRUB parser is not currently being worked on. But I'm wondering now if, just for Fedora 28-32, CentOS+RHEL 8, we want to implement a special case in the BLS parser just for this.

While I check out a special case, the following worked on **RHEL 8**, but I have not tried it on CentOS 8 -- you can switch to the actual systemd-boot with the following:

```bash
bootctl install
dnf remove grub2*
rm -rf /boot/loader # do this so bootctl and kernel-install won't assume these entries are already there
kernel-install add $(uname -r) /lib/modules/$(uname -r)/vmlinuz
```

hugelgupf, 2020-09-17:
This may be coming back to me -- I vaguely remember looking into just reading /grubenv and interpreting BLS entries with those variables in mind, but it turned out that while the BLS entries are in the EFI partition, the /grubenv is on the root partition of the OS. I was feeling a little uncomfortable with the cross-partition nature of that deduction/special case.

(I don't remember if this memory is true. I need to boot up my old RHEL 8 image and see what's going on.)

hugelgupf, 2020-09-17:
> This may be coming back to me -- I vaguely remember looking into just reading /grubenv and interpreting BLS entries with those variables in mind, but it turned out that while the BLS entries are in [...]

This looks to not be true in the files you provided, so maybe it's not that crazy. (In the files you provided, the grubenv is in EFI/centos/grubenv.)

gabhuang, 2020-09-17:
> This may be coming back to me -- I vaguely remember looking into just reading /grubenv and interpreting BLS entries with those variables in mind, but it turned out that while the BLS entries are in [...]
What I see is grubenv in EFI partition , and bls in root/loaders/entries, but there's a symbolic link to grubenv in root/grub2/
So they are not in the same partition :(

hugelgupf, 2020-09-17:
Ah, sadness. Does the workaround I posted work on CentOS for you? You'd have to boot the image somewhere else first, and then run those commands to generate spec-compliant BLS files. (Or boot something else, and manually edit the BLS spec to include the root= rather than $kernelopts, and then reboot into the image to generate real BLS files.)

gabhuang, 2020-09-17:
BLS files looks better with the workaround, I'll also try that on our platform.