u-root/u-root #1838: cpio: Zero-length file incorrectly treated as a hard link

mato, 2020-09-18:
After updating `u-root` from 382bc373 (November last year) to latest master (7a5b5be) I've found what looks like a bug in `cpio`.

----

Using the following shell script to create a minimal test case:

```
#!/bin/sh

set -xe

OUT=$(mktemp --suffix=.cpio)
TESTDIR=$(mktemp -d)
cd $TESTDIR

mkdir dir
touch dir/file
find . -mindepth 1 | cpio -H newc -o -F $OUT -R +1:+1 --reproducible

touch file
echo file | cpio -H newc -oA -F $OUT -R +0:+0 --reproducible
```

For reference, the output of GNU cpio `-it --verbose` on the test case:

```
drwxr-xr-x   2 daemon   daemon          0 Sep 18 13:37 dir
-rw-r--r--   1 daemon   daemon          0 Sep 18 13:37 dir/file
-rw-r--r--   1 root     root            0 Sep 18 13:37 file
1 block
```

----

With 382bc373 I get what I assumed was the correct behaviour, i.e. the last `file` is created as a zero-length file owned by `0:0`:

```
pread64(0, "file\0", 5, 346)            = 5
newfstatat(AT_FDCWD, ".", {st_mode=S_IFDIR|0755, st_size=17, ...}, 0) = 0
openat(AT_FDCWD, "file", O_RDWR|O_CREAT|O_TRUNC|O_CLOEXEC, 0666) = 3
```

with 7a5b5be `cpio` appears to think `file` is a hard link to `dir`, and fails with "link dir file: operation not permitted":

```
pread64(0, "file\0", 5, 346)            = 5
linkat(AT_FDCWD, "dir", AT_FDCWD, "file", 0) = -1 EPERM (Operation not permitted)
```

mato, 2020-09-18:
For completeness:

1. Before you ask "Why are you creating such a weird CPIO archive?", I'm using the above method as a round-about way to specify explicit ownership (1:1 vs 0:0) in a CPIO archive, without requiring the overall build to run as root. Blame UNIX tools for not having a sensible way to explicitly specify ownership in archives.
2. If the last `file` is not zero-length, then it all works fine on `master`; that's the workaround I'm currently employing.

rjoleary, 2021-05-27:
Some activity happened to the cpio command since. Specifically these two:

* https://github.com/u-root/u-root/pull/1898
* https://github.com/u-root/u-root/pull/1895

Can you check to see if it's better? Otherwise @rminnich  might be a good person to answer these questions.