u-root/u-root #1849: can't build external go commands into u-root with modules enabled (I suspect we know this?)

rminnich, 2020-10-07:
```
rminnich@rminnich-MacBookPro:~/mainboards/intel/hw$ make cpukernel
u-root -o cpu.cpio -build=bb -initcmd=cpud -files ~/.ssh/cpu_rsa.pub:key.pub \
	-defaultsh="" \
	github.com/u-root/cpu/cmds/cpud
2020/10/07 10:47:17 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/home/rminnich/go12 GOPATH=/home/rminnich/go CGO_ENABLED=0
2020/10/07 10:47:17 Filename is cpu.cpio
2020/10/07 10:47:18 error building: error building go package in "/home/rminnich/go/src/github.com/u-root/u-root/bb": go: finding module for package github.com/u-root/cpu/cmds/cpud/.bb
main.go:14:2: malformed module path "github.com/u-root/cpu/cmds/cpud/.bb": leading dot in path element
, exit status 1
Makefile:201: recipe for target 'cpu.cpio.lzma' failed
make: *** [cpu.cpio.lzma] Error 1
```

the "leading ." thing is confusing. 

This does work, thank goodness.
```
GO111MODULE=off make cpukernel
u-root -o cpu.cpio -build=bb -initcmd=cpud -files ~/.ssh/cpu_rsa.pub:key.pub \
	-defaultsh="" \
	github.com/u-root/cpu/cmds/cpud
2020/10/07 10:50:41 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/home/rminnich/go12 GOPATH=/home/rminnich/go CGO_ENABLED=0 GO111MODULE=off
2020/10/07 10:50:51 Successfully built "cpu.cpio" (size 4050616).
```

hugelgupf, 2020-10-07:
Go module support is still in progress. Sorry for the wait!

flokli, 2021-03-04:
Is there some tracking issue/wip branch for this? I tried [banging my head against getting a build pipeline for GitHub Actions](https://github.com/flokli/u-root/pull/1), but even in the unit tests there's various places where there's some assumptions on how the go source files are made available, making this pretty tough.

hugelgupf, 2024-02-20:
https://github.com/u-root/mkuimage is the new builder receiving attention, and there is some documentation on multi-module builds in the new README there.

closed 2024-02-20 by hugelgupf