u-root/u-root #1851: Inconsistent versions when getting master with modules enabled

gmacd, 2020-10-10:
If I run the steps below, I get an error about inconsistent versions:
```
mkdir test
cd test
go mod init test
go get -u github.com/u-root/u-root@master
```

Gives me the error:
```
go: github.com/u-root/u-root master => v1.0.1-0.20201001202801-7709c8272b97
go: golang.org/x/tools upgrade => v0.0.0-20201010145503-6e5c6d77ddcc
go: golang.org/x/xerrors upgrade => v0.0.0-20200804184101-5ec99f83aff1
go: golang.org/x/sys upgrade => v0.0.0-20201009025420-dfb3f7c4e634
go: golang.org/x/mod upgrade => v0.3.0
go get: inconsistent versions:
	github.com/u-root/u-root@v1.0.1-0.20201001202801-7709c8272b97 from github.com/u-root/u-root@master requires github.com/u-root/u-root@v6.0.1-0.20200118052101-6bcd1cda5996+incompatible (not github.com/u-root/u-root@v1.0.1-0.20201001202801-7709c8272b97 from github.com/u-root/u-root@master)
```

hugelgupf, 2020-10-10:
I think the `-u` is forcing an upgrade in dependencies, which is inconsistent with what you want.

hugelgupf, 2020-10-10:
Essentially, you should use `go get github.com/u-root/u-root@master` which will get u-root@master and the dependencies listed in the go.mod at master.

If you add `-u`, you get all that, *and* you try to upgrade all dependencies in the go.mod to a state we do not support.

gmacd, 2020-10-10:
Dropping -u fails faster:
```
go: github.com/u-root/u-root master => v1.0.1-0.20201001202801-7709c8272b97
go get: inconsistent versions:
	github.com/u-root/u-root@v1.0.1-0.20201001202801-7709c8272b97 from github.com/u-root/u-root@master requires github.com/u-root/u-root@v6.0.1-0.20200118052101-6bcd1cda5996+incompatible (not github.com/u-root/u-root@v1.0.1-0.20201001202801-7709c8272b97 from github.com/u-root/u-root@master)
```

gmacd, 2020-10-10:
I'm no go modules expert, but I suspect the dependency on iscinl might be causing problems, since it also refers to a specific older version of u-root.  (The same version mentioned above: `github.com/u-root/u-root v6.0.1-0.20200118052101-6bcd1cda5996+incompatible`)

hugelgupf, 2020-10-10:
Interesting. If you just `go get github.com/u-root/u-root`, you get version v7.0.0+incompatible. But @master gets you a pseudoversion of v1.0.1-0.2020.... at the moment, despite v7 being the newer version. (See https://golang.org/ref/mod#pseudo-versions).

I think this might be due to the missing version suffix on the module path -- i.e. not being github.com/u-root/u-root/v7. The last recognized valid last version is then v1.0.0 (despite all that pre-dating Go modules), and hence it generates v1.0.1.

We either need the /v7 suffix now, or we go delete all the git tags of the old non-Go-module compatible versions. Or both.

hugelgupf, 2020-10-10:
It's weird that some Go commands recognize v7.0.0 as the "real version", like go get, but the pseudo-version is based on the last valid "real version" of v1.0.0. That feels inconsistent.

gmacd, 2020-10-12:
Any thoughts as to the solution?  I'm afraid I don't have any good suggestions, but it does seem to block me from being able to use u-root via go modules and get the latest changes.

hugelgupf, 2020-10-14:
Hey, I got Ryan and Ron to agree that we should blow away our existing Go-module-incompatible release tags out of the Git repo.

Some of our dependencies currently depend on those existing version numbers, despite them being broken. Unfortunately, we're not in a position to just turn on Go modules for everything yet, or we'd probably just add the /v7 suffix.

I'm going to try it out in a forked repo somewhere first and see what breaks in dependencies, then push it here.

hugelgupf, 2020-10-14:
I suspect those dependencies are currently just the ones I've updated (iscsinl, p9, dhcp), most of which constitute that diamond dependency path your import is complaining about. So no way around it really anyway.

gmacd, 2020-12-19:
Hi - were you able to try removing the go-module-incompatible release tags?

closed 2024-02-27 by hugelgupf