u-root/u-root #1859: malformed module path "github.com/u-root/u-root/cmds/core/*": invalid char '*'

advancedwebdeveloper, 2020-10-18:
Hi.

I got such an issue:
> $ ./u-root
> 2020/10/18 18:48:56 Disabling CGO for u-root...
> 2020/10/18 18:48:56 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/home/oceanfish81/gollvm_dist GOPATH=/home/oceanfish81/go CGO_ENABLED=0 GO111MODULE=off
> 2020/10/18 18:48:57 WARNING: You are not using one of the recommended Go versions (have = go1.15.2, recommended = [go1.13 go1.14]).
> 			Some packages may not compile.
> 			Go to https://golang.org/doc/install to find out how to install a newer version of Go,
> 			or use https://godoc.org/golang.org/dl/go1.13 to install an additional version of Go.
> 2020/10/18 18:48:57 "github.com/u-root/u-root/cmds/core/*" is neither package or path/glob: malformed module path "github.com/u-root/u-root/cmds/core/*": invalid char '*'
> 

I am on Ubuntu 20.04.1:

> $ go env
> go verGO111MODULE=""
> GOARCH="amd64"
> GOBIN=""
> GOCACHE="/home/oceanfish81/.cache/go-build"
> GOENV="/home/oceanfish81/.config/go/env"
> GOEXE=""
> GOFLAGS=""
> GOHOSTARCH="amd64"
> GOHOSTOS="linux"
> GOINSECURE=""
> GOMODCACHE="/home/oceanfish81/go/pkg/mod"
> GONOPROXY=""
> GONOSUMDB=""
> GOOS="linux"
> GOPATH="/home/oceanfish81/go"
> GOPRIVATE=""
> GOPROXY="https://proxy.golang.org,direct"
> GOROOT="/home/oceanfish81/gollvm_dist"
> GOSUMDB="sum.golang.org"
> GOTMPDIR=""
> GOTOOLDIR="/home/oceanfish81/gollvm_dist/tools"
> GCCGO="/home/oceanfish81/gollvm_dist/bin/llvm-goc"
> AR="ar"
> CC="/usr/bin/clang"
> CXX="/usr/bin/clang++"
> CGO_ENABLED="1"
> GOMOD=""
> CGO_CFLAGS="-g -O2"
> CGO_CPPFLAGS=""
> CGO_CXXFLAGS="-g -O2"
> CGO_FFLAGS="-g -O2"
> CGO_LDFLAGS="-g -O2"
> PKG_CONFIG="pkg-config"
> GOGCCFLAGS="-fPIC -m64 -pthread -fno-caret-diagnostics -Qunused-arguments -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build248439025=/tmp/go-build -gno-record-gcc-switches -funwind-tables"
> 
.
I am using

> $ go version
> go version go1.15.2 gollvm LLVM 12.0.0git linux/amd64

rjoleary, 2020-10-28:
Ignore the "WARNING: You are not using one of the recommended Go versions" message. I sent out a PR to fix it.

I can't repro the issue you're having. Perhaps it's an issue with Go modules. Maybe @hugelgupf has an idea.

hugelgupf, 2020-10-29:
I'll look into this specific error, but I will say that both u-root and some of its dependencies have Go assembly in them, which will not be compatible with gollvm. It's unlikely that we'll support gollvm in general in the future.

hansinator, 2021-04-01:
Hey I'm having the same problem using
  go version go1.16.2 linux/amd64
Did anyone find a solution?

advancedwebdeveloper, 2021-04-03:
I am heading to reproduce for 

> $ go version
> go version go1.16.3 linux/arm64

advancedwebdeveloper, 2021-04-03:
@hugelgupf , I have the same issue on Aarch64:

> $ ./u-root
> ./u-root
> 2021/04/03 14:53:39 Disabling CGO for u-root...
> 2021/04/03 14:53:39 Build environment: GOARCH=arm64 GOOS=linux GOROOT=/usr/local/go GOPATH=/home/oceanfish81/go CGO_ENABLED=0 GO111MODULE=off
> 2021/04/03 14:53:39 WARNING: You are not using one of the recommended Go versions (have = go1.16.3, recommended = [go1.13 go1.14 go1.15]).
>                         Some packages may not compile.
>                         Go to https://golang.org/doc/install to find out how to install a newer version of Go,
>                         or use https://godoc.org/golang.org/dl/go1.13 to install an additional version of Go.
> 2021/04/03 14:53:39 "github.com/u-root/u-root/cmds/core/*" is neither package or path/glob: malformed import path "github.com/u-root/u-root/cmds/core/*": invalid char '*'
> 

CC @vielmetti , in case of any interest

vielmetti, 2021-04-05:
@advancedwebdeveloper "WARNING: You are not using one of the recommended Go versions". Can you try again with a supported version of Go.

advancedwebdeveloper, 2021-04-06:
Which versions of Golang would you propose?

hansinator, 2021-04-08:
Works with go 1.14

joonas-fi, 2021-04-08:
Can confirm, works with Go 1.14 but not with 1.16. The problem is related to Go modules IIUC.

This works:

```console
$ docker run --rm -it golang:1.14 bash
$ go get github.com/u-root/u-root
$ u-root
2021/04/08 16:53:55 Disabling CGO for u-root...
2021/04/08 16:53:55 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/go CGO_ENABLED=0 GO111MODULE=off
2021/04/08 16:53:55 Skipping package "/go/src/github.com/u-root/u-root/cmds/core/bind": no buildable Go source files in /go/src/github.com/u-root/u-root/cmds/core/bind
2021/04/08 16:54:07 Successfully built "/tmp/initramfs.linux_amd64.cpio" (size 14929060).
```

alishir, 2021-05-24:
Works with Go 1.14 but does not work with Go 1.14.15.

rjoleary, 2021-06-07:
Can you try with `export GO111MODULE=off`?

rgl, 2021-07-30:
> Can you try with `export GO111MODULE=off`?

It fails with:

```console
vagrant@ovmf:~$ go version
go version go1.16.6 linux/amd64
vagrant@ovmf:~$ export GO111MODULE=off
vagrant@ovmf:~$ u-root
2021/07/30 14:20:19 Disabling CGO for u-root...
2021/07/30 14:20:19 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/home/vagrant/go CGO_ENABLED=0
2021/07/30 14:20:19 "github.com/u-root/u-root/cmds/core/*" is neither package or path/glob: cannot find package "github.com/u-root/u-root/cmds/core/*" in any of:
	/usr/local/go/src/github.com/u-root/u-root/cmds/core/* (from $GOROOT)
	/home/vagrant/go/src/github.com/u-root/u-root/cmds/core/* (from $GOPATH)
```

rgl, 2021-10-06:
ping?

dexter-ampere, 2021-11-23:
Dockerfile reproducer can be found [here](https://gist.github.com/dexter-ampere/f0f5a5c752ecfcd1bc9e51d18b216795)

[image]

[image]

1) `docker build -t u-root.malformed.img .`
2) Notice GO111MODULE not set in the Dockerfile but put in the .bashrc for later use
3) `docker run --name u-root.malformed.cntnr -it --rm u-root.malformed.img bash`
4) Type `u-root` and notice the error below _**note: this is u-root that Dockerfile built**_

```
[root@e0b6d2dc5c27 ~]# u-root
2021/11/23 00:02:45 Disabling CGO for u-root...
2021/11/23 00:02:45 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib/golang GOPATH=/root/go CGO_ENABLED=0
2021/11/23 00:02:45 WARNING: You are not using one of the recommended Go versions (have = go1.16.7, recommended = [go1.13 go1.14]).
                        Some packages may not compile.
                        Go to https://golang.org/doc/install to find out how to install a newer version of Go,
                        or use https://godoc.org/golang.org/dl/go1.13 to install an additional version of Go.
2021/11/23 00:02:45 Filename is /tmp/initramfs.linux_amd64.cpio
2021/11/23 00:02:45 "github.com/u-root/u-root/cmds/core/*" is neither package or path/glob: malformed import path "github.com/u-root/u-root/cmds/core/*": invalid char '*'
```

5) Run `go get github.com/u-root/u-root` again to rebuild

6) Type `u-root` and notice that it now works **_note: GO111MODULE is set in .bashrc so step no. 3 used it_**

```
2021/11/23 00:28:20 Disabling CGO for u-root...
2021/11/23 00:28:20 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib/golang GOPATH=/root/go CGO_ENABLED=0
2021/11/23 00:28:20 WARNING: You are not using one of the recommended Go versions (have = go1.16.7, recommended = [go1.17]).
                        Some packages may not compile.
                        Go to https://golang.org/doc/install to find out how to install a newer version of Go,
                        or use https://godoc.org/golang.org/dl/go1.17 to install an additional version of Go.
2021/11/23 00:28:20 Skipping package "/root/go/src/github.com/u-root/u-root/cmds/core/bind": no buildable Go source files in /root/go/src/github.com/u-root/u-root/cmds/core/bind
2021/11/23 00:28:20 Skipping package "/root/go/src/github.com/u-root/u-root/cmds/core/chroot": //go:build comment without // +build comment
2021/11/23 00:28:20 Skipping package "/root/go/src/github.com/u-root/u-root/cmds/core/df": //go:build comment without // +build comment
2021/11/23 00:28:20 Skipping package "/root/go/src/github.com/u-root/u-root/cmds/core/dmesg": //go:build comment without // +build comment
2021/11/23 00:28:34 Successfully built "/tmp/initramfs.linux_amd64.cpio" (size 13073308).
```

hugelgupf, 2024-02-20:
It's been a while and the build system has seen a lot of progress. I believe this should be resolved. Please reopen / file another bug if not.

closed 2024-02-20 by hugelgupf