u-root/u-root #1863: Documentation: examples using curly braces globs don't work in practice.

uucidl, 2020-10-25:
The README suggests using {a, b, c} to install commands from the exp/ category. This however doesn't work because go's filepath package does not support curly braces globbing, as shown here: https://golang.org/pkg/path/filepath/#Match

rjoleary, 2020-10-28:
It still works if you run from a shell like bash or zsh. It's not the u-root command doing the expansion, but rather the shell:

```
% echo cmds/core/{init,ls,ip,dhclient,wget,cat,elvish}
cmds/core/init cmds/core/ls cmds/core/ip cmds/core/dhclient cmds/core/wget cmds/core/cat cmds/core/elvish
```

uucidl, 2020-11-01:
The reason I reported it as a documentation bug was that if you rely on shell expansion, you can only run the `u-root` at the top of the `u-root` tree.

This isn't so natural when installing with go get, where you are tempted to simply run `u-root` from anywhere

rjoleary, 2021-05-27:
Do you know if there's an existing Go function to expand the curly brackets? Something like:

`func ExpandBrackets(input string) []string`

It should be pretty easy to write, but I feel something like this might already exist in the standard library.

uucidl, 2021-05-27:
Sorry I don't know enough the go standard library. I did search for it back then when filing the issue, and did not find a ready-made function to expand the bracket expressions

rjoleary, 2021-07-03:
We ran into this problem in the LinuxBoot build system too. Make doesn't seem to expand curly braces.

hugelgupf, 2024-02-26:
The latest version of gobusybox supports shell expansions in the Go API, including curly braces. #2923 should update to it.

closed 2024-02-27 by hugelgupf