u-root/u-root #1871: Add support for blscfg in Grub parser`

insomniacslk, 2020-11-05:
`blscfg` allows using Grub to redirect to BLS for the boot configuration. We want to extend our "parser" to support scenarios like this. This requires support for flow control in Grub, hence some form of real parsing other than regexps like we currently do. This will be a bloodbath.

rjoleary, 2021-05-27:
Would this allow us to boot CentOS 8?

rjoleary, 2021-05-27:
There's the shell of a recursive decent parser here: https://github.com/u-root/u-root/tree/master/pkg/pogosh. A similar idea to this could be used for grub parsing.

rjoleary, 2021-05-27:
Might be related to https://github.com/u-root/u-root/issues/1835

hugelgupf, 2024-02-20:
We support this now, though it's probably a bit hacked up.

closed 2024-02-20 by hugelgupf