u-root/u-root #1875: Can't boot ubuntu 20.04 anymore

vejmarie, 2020-11-09:
It used to work in the past, but with a freshly installed system (installer boot works) on sda1 I am receiving this

2020/11/09 21:45:36 [grub] Got config file file:///tmp/u-root-mounts963802198/sda1/EFI/ubuntu/grub.cfg:
search.fs_uuid be271b43-e570-4af7-9c9a-f9315636679d root hd0,gpt2
set prefix=($root)'/boot/grub'
configfile $prefix/grub.cfg

2020/11/09 21:45:36 Warning: Grub parser could not parse ["search" "--fs-uuid" "be271b43-e570-4af7-9c9a-f9315636679d" "root" "hd0,gpt2"]
2020/11/09 21:45:36 No GRUB configs found on BlockDevice(name=sda1, fs_uuid=3f72-d409), trying another format...: no valid grub config found
2020/11/09 21:45:36 No syslinux configs found on BlockDevice(name=sda1, fs_uuid=3f72-d409): no valid syslinux config found on /tmp/u-root-mounts963802198/sda1

rjoleary, 2020-11-09:
That syntax for `search.fs_uuid` doesn't match https://www.gnu.org/software/grub/manual/grub/html_node/search.html . 

Perhaps we need a `continue` after this line: https://github.com/u-root/u-root/blob/master/pkg/boot/grub/grub.go#L351

vejmarie, 2020-11-09:
You mean it has too many parameters ?

hugelgupf, 2020-11-09:
I'm gonna say -- it doesn't matter much what the documentation states if it's wrong. What matters is that it works -- which should be our compatibility target

vejmarie, 2020-11-09:
> I'm gonna say -- it doesn't matter much what the documentation states if it's wrong. What matters is that it works -- which should be our compatibility target

Agree with you. I will try to debug that tonight. What surprised me is that this install was working like a charm a couple of months ago and it is now broken. Looks like that we need to ramp up "live" testing which is another story.

hugelgupf, 2020-11-10:
> Agree with you. I will try to debug that tonight. What surprised me is that this install was working like a charm a couple of months ago and it is now broken. Looks like that we need to ramp up [...]

We previously ignored the `search` command altogether. Ryan recently added support. https://github.com/u-root/u-root/pull/1840

vejmarie, 2020-11-10:
Ok, with the continue as suggested by @rjoleary , the entry is "properly" detected and Ubuntu is available as a boot option and starts. I just "recovered" my OCP demo 3 hours ahead of it that is not bad. 
I think the end of the command shall be considered as --hint. What do you want to do longer term, parse it as hint, or just get ride of it ?

rjoleary, 2020-11-19:
The `continue` is a good fix. It essentially means that if there is a parse error, ignore the line and keep going.

closed 2021-05-25 by rjoleary