u-root/u-root #1886: Remove pclntab

xaionaro, 2020-12-14:
This is a spin-off of discussion: https://github.com/u-root/u-root/issues/1477

It was investigated that the `pclntab` might take 10-40% of the binary size. And I proposed in https://github.com/golang/go/issues/36555 to add an option to strip the `pclntab` off for "embedded use cases". It seems they did not like the idea, however there exists a derivative golang project "[embeddedgo](https://github.com/embeddedgo/go)", which implemented this option.

I'm wondering if it makes sense to try that out? Or we want to use the upstream version of the compiler no matter what? :)

rjoleary, 2021-05-27:
I think this is worth considering. 10-40% is really big.

This wouldn't really affect u-root right? The version of Go used is whichever the user has in their $PATH. Users who don't care will just use the upstream compiler. Users that do care can build their own.

Could it be apart of the docker image https://github.com/u-root/u-root/tree/master/tools/golang_patched_dce ?

hugelgupf, 2021-05-27:
Go 1.16 and 1.17 have actually made a few binary size improvements. We
should re-measure and shelve for a lil while?

xaionaro, 2021-05-31:
> Go 1.16 and 1.17 have actually made a few binary size improvements.

In the discussion about pclntab they only were going to compress it. But we compress u-root in initrd anyway, so it should not help us much.

> Could it be apart of the docker image https://github.com/u-root/u-root/tree/master/tools/golang_patched_dce ?

Uh, nice idea.