u-root/u-root #1894: Busybox mode?

pwFoo, 2020-12-24:
Hi,
started playing with u-root again and fail to build a bb mode binary? Tried without option (read bb mode should be default?) and with `-build=bb` or `-format=bb`. Also tried with space between... Only get a `/tmp/initramfs.linux_amd64.cpio` and no bb binary?

Tried today with actual source of u-root.

rminnich, 2020-12-30:
can you show
command you typed
cpio -ivt < /tmp/initramfs*cpio

pwFoo, 2020-12-30:
Hi @rminnich,
I try to build a binary instead of Initrd. Or do I need to extract the binary?

pwFoo, 2021-01-02:
Hi @rminnich I'm able to extract the bb binary frpm cpio, but still not understand way I need to build an initramfs and extract the binary in a second step? I know there is (or was?!) an `-format bb` to generate an binary file instead of an cpio...?

hugelgupf, 2021-01-02:
The tool in the directory `tools/makebb` can generate just the binary for you

pwFoo, 2021-01-02:
@hugelgupf 
Thanks! Looks like the `-format bb` is removed from u-root. makebb works fine to build a default cmd set or custom builds too
```
/go/bin/makebb github.com/u-root/u-root/cmds/core/cp github.com/u-root/u-root/cmds/core/mv github.com/u-root/u-root/cmds/core/ps github.com/u-root/u-root/cmds/core/mount 
```

hugelgupf, 2021-01-02:
-format=bb has always produced an initramfs, it hasn't been removed.

pwFoo, 2021-01-02:
Strange... I I thought I had used that in the past... But not sure anymore...
makebb works fine for my testing use case! 

Optional could be added as format bb additional to cpio?

makebb solved it for me

closed 2021-05-27 by rjoleary