u-root/u-root #1904: Support for running a LDD-like command against cross-compiled binaries?

ChrisTrenkamp, 2021-01-07:
I've been hacking the u-root code to try to get it to run a LDD-like command against cross-compiled binaries.  My use-case is I want to build an initramfs with Gentoo's `crossdev` environment.  

`crossdev` builds a GCC cross-compiler and a new `emerge` location for building your packages.  For example, if you want to build a new environment for an Arm64 device, run `crossdev aarch64-unknown-linux-gnu`, and it'll create the cross compilers.  When you run `aarch64-unknown-linux-gnu-emerge iw`, it'll build the `iw` package and put the files under `/usr/aarch64-unknown-linux-gnu/`.

The problem with using u-root for building an initramfs with `crossdev` builds are:

* The standard `ld-linux.so` utility doesn't work against cross-compiled binaries.
* Even though the dynamic libaries are being placed under `/usr/*architecture here*/usr/lib/`, the dynamic links the executables look for is under `/usr/lib/`.

I found a script that simulates LDD for cross-compiled binaries: https://gist.github.com/jerome-pouiller/c403786c1394f53f44a3b61214489e6f .  I just had to copy this script to `/usr/bin/aarch64-unknown-linux-gnu-ldd` and it works with the `crossdev` compilers.

I then made some hacks to support `crossdev`.  Basically it strips paths with a specific path and runs a custom LDD command: https://github.com/ChrisTrenkamp/u-root/commit/e1a9ffea10282baf5e87da6626045547b60d3ea1

With all these changes, I ran this script to build an initramfs:

```bash
export GOARCH="arm64" GOOS="linux"

u-root \
  -o initramfs.cpio \
  -files \
    /usr/aarch64-unknown-linux-gnu/usr/sbin/iw:bin/iw \
  -ldd-base /usr/aarch64-unknown-linux-gnu/ \
  -ldd-binary /usr/bin/aarch64-unknown-linux-gnu-ldd
```

And here's a test run in a VM:

```
qemu-system-aarch64 -M virt -m 1024M -cpu cortex-a53 -kernel debian-linux -nographic -initrd initramfs.cpio
# Kernel init logs...
2021/01/07 14:53:27 Welcome to u-root!
                              _
   _   _      _ __ ___   ___ | |_
  | | | |____| '__/ _ \ / _ \| __|
  | |_| |____| | | (_) | (_) | |_
   \__,_|    |_|  \___/ \___/ \__|

# Have to set the LD_LIBRARY_PATH since by default it only looks under /lib64 and /usr/liib64
E:LD_LIBRARY_PATH="/lib:/lib64:/usr/lib:/usr/lib64"
iw
Usage:	/bin/iw [options] command
Options:
	--debug		enable netlink debugging
	--version	show version (5.9)
...
```

It indeed works.  Since u-root has support for building initramfs' for different architectures, should it also support copying dependencies for cross-compiled binaries?

flokli, 2021-03-04:
Do you really need ldd here? `ldd` [executes](http://klamp.works/2016/04/15/code-exec-ldd.html) the binary you're investigating. It doesn't respect the interpreter listed in the binary to execute. It considers `LD_LIBRARY_PATH` being set when running `ldd`. Doing this on a binary for another architecture can't work.

Shouldn't `readelf` suffice?

rjoleary, 2021-05-27:
This is something I'd like to see too. I've run into this problem a few times when cross-compiling.

@flokli For the u-root command, we have our own ldd implementation which seems to just parse the elf file: https://github.com/u-root/u-root/tree/master/pkg/ldd