u-root/u-root #1919: -files misbehaves when specifying a directory with `/bbin/bb` in the tree

synackd, 2021-01-25:
To illustrate this, consider the following minimal example:

Assume that `"$TMPDIR"/base` contains:

```
"$TMPDIR"/base/
└── bbin
    ├── bb
    ├── elvish -> bb
    └── init -> bb
```

`bb` was generated by [gobusybox](https://github.com/u-root/gobusybox) and the symlinks by `ln -s`. Now, to generate an initramfs, run:

```shell
$ u-root -nocmd -files "$TMPDIR"/base:. -o initramfs.cpio
```

The expected behavior is that `/bbin/bb`, `/bbin/elvish`, and `/bbin/init` should be present in `initramfs.cpio` along with the other files u-root generates, e.g. `/dev/console`.

The actual behavior is that the files mentioned above are indeed included in the correct places in `initramfs.cpio`, but `"$TMPDIR"/base/bbin/bb` is also included when it shouldn't be.

synackd, 2021-01-26:
After some further digging, it seems that u-root is trying to add the `ldd` dependencies of the `/bbin/bb` it adds to the archive, so the _original_ path (`"$TMPDIR"/base/bbin/bb`) is included a bunch of times.

This occurs [here](https://github.com/u-root/u-root/blob/fd79f5810d4ffc6b6a33357af42b94c2b3bd6b9f/pkg/uroot/uroot.go#L512) in `pkg/uroot/uroot.go`.

Should a command-line option be added or autodetection occur to disable adding `ldd` dependencies for the BusyBox binary for situations where gobusybox was used to create it?

synackd, 2021-02-08:
After further testing, it seems that ldd already successfully detects that `bb` is statically linked (to answer @GanShun's question [here](https://github.com/u-root/u-root/pull/1931#pullrequestreview-584400818)). The issue occurs when there are symlinks to `bb` (e.g. a program name like `echo -> bb`), since ldd always follows symlinks and adds them as absolute paths.

Rather than alter this behavior, I think this can be remedied by doing one or both of two things:

1. Allowing the `u-root` command to allow reading from and writing to pipes. That way one could do something like:
```bash
# Add bb and its symlinks (in basedir/) to a u-root initramfs.
# u-root will add the necessary device files, etc.
( cd basedir/ && find . | cpio -oc | u-root -nocmd ) > initramfs.cpio
```

2. Modifying `-base` to include the files on top of the base cpio that are normally created by u-root, i.e.
```
bin
dev
dev/console
dev/null
dev/port
dev/tty
dev/urandom
env
etc
etc/localtime
etc/resolv.conf
lib64
proc
sys
tcz
tmp
ubin
usr
usr/lib
var
var/log
```

@rminnich, what do you think of these?

hugelgupf, 2024-02-20:
I'd probably use the -skip-ldd cmdline option. I agree the API isn't the nicest in this case, however.