u-root/u-root #1954: Running `go test ./...` shouldn't fail

flokli, 2021-03-04:
I cloned the repo, and ran the tests via `go test ./...`. Some of them seem to fail.

I further restricted this to `go test -v ./pkg/boot/...`, and there still seem to be some failures.

There's a lot of other prints going on, so it's hard to see what's failing.

One failure seems to be:
```
    kver_test.go:119: mismatch
    kver_test.go:122: 
        want bzimage.KInfo{Release:"2.6.24.111", Version:"#606 Mon Apr 14 00:06:11 CEST 2014", Builder:"", BuildNum:0x25e, BuildTime:time.Time{wall:0x0, ext:63533030771, loc:(*time.Location)(nil)}, Maj:0x2, Min:0x6, Patch:0x18, LocalVer:""}
        got  bzimage.KInfo{Release:"2.6.24.111", Version:"#606 Mon Apr 14 00:06:11 CEST 2014", Builder:"", BuildNum:0x25e, BuildTime:time.Time{wall:0x0, ext:63533023571, loc:(*time.Location)(0x6bf8c0)}, Maj:0x2, Min:0x6, Patch:0x18, LocalVer:""}    
--- FAIL: TestParseDesc (0.00s)
    --- PASS: TestParseDesc/normboot (0.00s)
    --- FAIL: TestParseDesc/ancient (0.00s)
FAIL
FAIL  github.com/u-root/u-root/pkg/boot/bzimage 11.944s
```

I'd feel more comfortable adding changes if `go test ./...` would succeed, and CI would uncover/prevent `go test` from regressing.

hugelgupf, 2021-03-04:
go test is part of CI.

What version of Go are you on?

hugelgupf, 2021-03-04:
See https://github.com/u-root/u-root/blob/859ca5728ec336bced14fb3c3b3aaa7b94bc907f/.circleci/config.yml#L125

flokli, 2021-03-04:
I don't have 1.13 provided anymore in my distribution. Should this line succeed with other Go versions, like 1.15 or 1.16 too?

Further looking at the `config.yml`, where is `uroottest/test-image-amd64:v3.2.13` defined?

hugelgupf, 2021-03-04:
> I don't have 1.13 provided anymore in my distribution. Should this line succeed with other Go versions, like 1.15 or 1.16 too?

1.15 def. I was just discovering some other 1.16 issues earlier, so I doubt 1.16 works right now.

> Further looking at the `config.yml`, where is `uroottest/test-image-amd64:v3.2.13` defined?

Dockerhub

hugelgupf, 2021-03-04:
Oh, sorry, definition for the image is in .circleci/images

flokli, 2021-03-04:
> 1.15 def. I was just discovering some other 1.16 issues earlier, so I doubt 1.16 works right now.

Yeah, I was just confused by the readme explicitly mentioning 1.13. 

Okay, running

```
go test -a -timeout 15m -ldflags '-s' ./cmds/... ./pkg/... 
```

certainly fails here:

[test.log](https://github.com/u-root/u-root/files/6082624/test.log)

rjoleary, 2021-05-27:
I don't believe we test with MacOS in the CI, so all bets are off there. Pull Requests to improve it will be accepted.

rjoleary, 2021-05-27:
A bunch of tests broke in go1.15+ and were fixed in https://github.com/u-root/u-root/pull/1974. Please try again.

Unfortunately, the CI only tests with go1.13 so these issues aren't caught.

rjoleary, 2021-05-27:
You might also need to set `GO111MODULE=off` when running the tests (as does the CI).

rjoleary, 2021-05-27:
So, I tried running `go test ./...` locally myself, it hanged and I can't CTRL-C it either...

closed 2024-02-26 by hugelgupf