u-root/u-root #1955: Go test shouldn't create files in the worktree

flokli, 2021-03-04:
After running unit tests, the following files/folders seem to have been created:

```
bb/
cmds/core/wget/200
cmds/core/wget/302
pkg/crypto/tests/private_key2.pem
pkg/crypto/tests/public_key2.pem
```

These should probably created in a tmpdir, and removed afterwards.

hugelgupf, 2021-03-04:
`bb` - will be changed when we switch to gobusybox, but for the moment you'll have to live with it. That's a hard one to fix.

The others, probably agreed.

flokli, 2021-03-04:
If we bump the requirement to Go 1.15, we could use `TempDir` from `testing` to change directory before running the wget binaries.

The pem files are produced by `pkg/crypto/ed25519_test.go`, which wouldn't require changing directories - but we can just pass the tempdir as a location to create them in.

hugelgupf, 2021-03-04:
I don't think we're ready to bump to Go 1.15. Some users' companies tend to lag a bit behind.

proposed solutions sound good to me

flokli, 2021-03-04:
I opened https://github.com/u-root/u-root/pull/1956, which addresses `ed25519_test.go`.

rjoleary, 2021-05-27:
Just curious, what is the issue with these files so long as they are in a .gitignore? It feels like putting all generated files in a tmpdir, then running `defer os.RemoveAll(tmpdir)` afterwards just makes the test harder to debug when it fails.

flokli, 2021-05-27:
There's nothing guaranteeing the files are created again. Files might already exist from a previous run, might be in an older version etc. They don't need to be around at all after the test run.

The Go compiler people also started chasing down file leakage, see https://github.com/golang/go/issues/35998 introducing `TB.TempDir()`, and the links in the initial description.