u-root/u-root #1959: Code coverage only covers folders with _test.go

chengchiehhuang, 2021-03-08:
I noticed that the current code coverage reported does not include folder with no _test.go. See https://app.codecov.io/gh/u-root/u-root/compare/1896/tree/pkg 

This creates problem I put new code and test in an old folder.... Are we supposed to include everything in the code coverage report?

hugelgupf, 2021-03-08:
Yep noticed this too this weekend. Imo we need to.

rjoleary, 2021-05-25:
Could we just put an empty "pkg_test.go" in each package without a test today?

hugelgupf, 2021-05-25:
I mean... Yes we can. But do you want those coverage numbers to drop rn? 😬

rjoleary, 2021-05-27:
I tested it out with the following (+ a few tweaks):

```
~/go/src$ for pkg in $(go list github.com/u-root/u-root/cmds/...); do if ! (ls $pkg | grep _test.go > /dev/null); then echo package main > $pkg/pkg_test.go; gofmt -w $pkg/pkg_test.go; fi; done
~/go/src$ for pkg in $(go list github.com/u-root/u-root/pkg/...); do if ! (ls $pkg | grep _test.go > /dev/null); then echo package $(basename $pkg) > $pkg/pkg_test.go; gofmt -w $pkg/pkg_test.go; fi; done
```

* Before: 36.4%
* After: 26.6%

So we get a 10% hit.

Also, for some reason, when I run locally, it's a lot less than in the CI. Possibly it's because some of the tests are failing locally. Possibly the CI is measuring statements instead of lines.

rjoleary, 2021-05-27:
Is the coverage drop acceptable? Otherwise, it seems this issue is moot. Either we cover everything and drop, or do nothing.

Third option is to wait until all these packages receive actual unit tests, and that would be quite the undertaking.

fotonick, 2021-06-02:
I personally want accuracy. Let's not anchor on incorrect numbers. That is, it's not a real drop, but a perceived drop.

hugelgupf, 2021-06-02:
Then I'm for accuracy.

Note that it will continue to be somewhat deceiving: Go test coverage is
unit test coverage.

A test in pkg/x *only* collects coverage about pkg/x, not about any other
packages used. Some of our tests (esp in pkg/boot subpackages) are more
integration-testy and cover multiple packages whose coverage won't be
measured as is. (It's worth untangling that and covering everything for
real, imo.)

We also still don't collect VM test coverage, which I had an open
unfinished PR for.

rjoleary, 2021-06-03:
We also have this issue in measuring coverage: https://github.com/u-root/u-root/issues/116

rjoleary, 2021-06-03:
But, I they're orthogonal -- we can fix this issue without waiting for those others.

closed 2021-11-24 by rjoleary