u-root/u-root #1970: ESXi boot stuck after loading image

yurinnick, 2021-03-24:
`esxiboot` loads all modules, but getting stuck right after loading kernel image with no output anywhere.

Hardware: HP DL360 G10

Output:
```
esxiboot -c ./esxi.cfg

2021/03/24 02:36:25 Parsing multiboot header
2021/03/24 02:36:25 VideoMode flag is not supproted yet, trying to load anyway
2021/03/24 02:36:25 Found multiboot image
2021/03/24 02:36:25 Getting kernel entry point
2021/03/24 02:36:25 Kernel entry point at 0x4117b0
2021/03/24 02:36:25 Parsing ELF segments
2021/03/24 02:36:25 Parsing memory map
2021/03/24 02:36:25 Preparing multiboot info
2021/03/24 02:36:25 Memory map: [0x0, 0xa0000) (len: 0xa0000, size: 0x14, type: 1)
[0xf0000, 0x100000) (len: 0x10000, size: 0x14, type: 2)
[0x100000, 0x71608000) (len: 0x71508000, size: 0x14, type: 1)
[0x71608000, 0x7160c000) (len: 0x4000, size: 0x14, type: 2)
[0x7160c000, 0x7161c000) (len: 0x10000, size: 0x14, type: 1)
[0x7161c000, 0x71620000) (len: 0x4000, size: 0x14, type: 2)
[0x71620000, 0x93e97000) (len: 0x22877000, size: 0x14, type: 1)
[0x93e97000, 0x957b4000) (len: 0x191d000, size: 0x14, type: 2)
[0x957b4000, 0x957b8000) (len: 0x4000, size: 0x14, type: 1)
[0x957b8000, 0x969fb000) (len: 0x1243000, size: 0x14, type: 2)
[0x969fb000, 0xa01fb000) (len: 0x9800000, size: 0x14, type: 1)
[0xa01fb000, 0xa0efb000) (len: 0xd00000, size: 0x14, type: 2)
[0xa0efb000, 0xa32fb000) (len: 0x2400000, size: 0x14, type: 4)
[0xa32fb000, 0xa34fb000) (len: 0x200000, size: 0x14, type: 3)
[0xa34fb000, 0xb0000000) (len: 0xcb05000, size: 0x14, type: 1)
[0xb0000000, 0xd0000000) (len: 0x20000000, size: 0x14, type: 2)
[0xfe000000, 0xfe010000) (len: 0x10000, size: 0x14, type: 2)
[0x100000000, 0xc03ffff000) (len: 0xbf3ffff000, size: 0x14, type: 1)
[0xc03ffff000, 0xc040000000) (len: 0x1000, size: 0x14, type: 2)

2021/03/24 02:36:26 Added module jumpstrt.gz at [0x601000, 0x601000)
2021/03/24 02:36:26 Added module useropts.gz at [0x601000, 0x601000)
2021/03/24 02:36:26 Added module features.gz at [0x601000, 0x601000)
2021/03/24 02:36:26 Added module k.b00 at [0x601000, 0x126cd42)
2021/03/24 02:36:26 Added module chardevs.b00 at [0x126d000, 0x127c524)
2021/03/24 02:36:26 Added module user.b00 at [0x127d000, 0x145dc04)
2021/03/24 02:36:26 Added module procfs.b00 at [0x145e000, 0x1464044)
2021/03/24 02:36:26 Added module uc_intel.b00 at [0x1465000, 0x15b8400)
2021/03/24 02:36:26 Added module uc_amd.b00 at [0x15b9000, 0x15babdc)
2021/03/24 02:36:26 Added module uc_hygon.b00 at [0x15bb000, 0x15bb01c)
2021/03/24 02:36:26 Added module vmx.v00 at [0x15bc000, 0x323eeb8)
2021/03/24 02:36:26 Added module vim.v00 at [0x323f000, 0x42e01d8)
2021/03/24 02:36:26 Added module sb.v00 at [0x42e1000, 0x6f1bb84)
2021/03/24 02:36:26 Added module s.v00 at [0x6f1c000, 0x7ac8ddc)
2021/03/24 02:36:26 Added module lsi_mr3.v00 at [0x7ac9000, 0x7b24a48)
2021/03/24 02:36:26 Added module bnxtnet.v00 at [0x7b25000, 0x7be2158)
2021/03/24 02:36:26 Added module bnxtroce.v00 at [0x7be3000, 0x7c46718)
2021/03/24 02:36:26 Added module elx_esx_.v00 at [0x7c47000, 0x82af6d1)
2021/03/24 02:36:26 Added module brcmfcoe.v00 at [0x82b0000, 0x84e7cd0)
2021/03/24 02:36:26 Added module elxiscsi.v00 at [0x84e8000, 0x8569990)
2021/03/24 02:36:26 Added module elxnet.v00 at [0x856a000, 0x860bac0)
2021/03/24 02:36:26 Added module lpfc.v00 at [0x860c000, 0x88dd46d)
2021/03/24 02:36:26 Added module bootcfg.v00 at [0x88de000, 0x88fd256)
2021/03/24 02:36:26 Added module conrep.v00 at [0x88fe000, 0x98458e2)
2021/03/24 02:36:26 Added module cru.v00 at [0x9846000, 0x98501b1)
2021/03/24 02:36:26 Added module fc_enabl.v00 at [0x9851000, 0xb773398)
2021/03/24 02:36:26 Added module hpessacl.v00 at [0xb774000, 0xcda4a1c)
2021/03/24 02:36:26 Added module hponcfg.v00 at [0xcda5000, 0xce21f08)
2021/03/24 02:36:26 Added module ilo.v00 at [0xce22000, 0xce2b1da)
2021/03/24 02:36:26 Added module ilorest.v00 at [0xce2c000, 0xd3d293a)
2021/03/24 02:36:26 Added module oem_buil.v00 at [0xd3d3000, 0xd3d502b)
2021/03/24 02:36:26 Added module scsi_hpd.v00 at [0xd3d6000, 0xd6e7130)
2021/03/24 02:36:26 Added module smx_prov.v00 at [0xd6e8000, 0x10f266e0)
2021/03/24 02:36:26 Added module sut.v00 at [0x10f27000, 0x1191a0dd)
[... 119 lines cut ...]
2021/03/24 02:36:26 Added module lsu_lsi_.v03 at [0x15e5e000, 0x15e9f12c)
2021/03/24 02:36:26 Added module lsu_lsi_.v04 at [0x15ea0000, 0x15f2b630)
2021/03/24 02:36:26 Added module lsu_smar.v00 at [0x15f2c000, 0x15f43684)
2021/03/24 02:36:26 Added module native_m.v00 at [0x15f44000, 0x16010898)
2021/03/24 02:36:26 Added module rste.v00 at [0x16011000, 0x160df7d0)
2021/03/24 02:36:26 Added module vmware_e.v00 at [0x160e0000, 0x1610e87c)
2021/03/24 02:36:26 Added module vsan.v00 at [0x1610f000, 0x18c60858)
2021/03/24 02:36:26 Added module vsanheal.v00 at [0x18c61000, 0x19407904)
2021/03/24 02:36:27 Added module vsanmgmt.v00 at [0x19408000, 0x1ad5158f)
2021/03/24 02:36:27 Added module tools.t00 at [0x1ad52000, 0x27291000)
2021/03/24 02:36:27 Added module xorg.v00 at [0x27291000, 0x275dab00)
2021/03/24 02:36:27 Added module imgdb.tgz at [0x275db000, 0x277ae800)
2021/03/24 02:36:27 Added module imgpayld.tgz at [0x277af000, 0x27dae000)
2021/03/24 02:36:27 Adding trampoline
2021/03/24 02:36:27 Trampoline entry point at 0x103000
2021/03/24 02:36:27 Loaded image: MultibootImage(
  Name: ESXi installer for HP from config file ./esxi.cfg
  Kernel: b.b00
  Cmdline: b.b00 runweasel addvmportgroup=0 mkernel.boot.tty2=com1 vmkBootVerbose=TRUE vmbLog=TRUE debugLogToSerial=1 logPort=com1
  iBFT: <nil>
  Modules: jumpstrt.gz, useropts.gz, features.gz, k.b00, chardevs.b00, user.b00, procfs.b00, uc_intel.b00, uc_amd.b00[  395.910286] kexec_core: Starting new kernel
, uc_hygon.b00,
```

Does it mean it doesn't get loaded due to missing video output or misconfigured serial port? What's the best way to debug it? Thanks!

hugelgupf, 2021-03-24:
Are you able to tell us what version the image is?

Please also note what version of Go you used to compile esxiboot. If you're on 1.16, please give 1.15 a try. 

The best way to debug would be to attempt to repro in qemu. We do have a test where you could plug in an ESXi *disk* image at https://github.com/u-root/u-root/blob/2bed9dbc8fe022811673cd64494963a32b1f56bd/integration/generic-tests/esxi_boot_test.go#L21 (Set the env var to a disk image to boot. You can probably easily modify the test to boot the way you do on a real machine if you pass the right stuff to qemu.)

General integration test guide is https://github.com/u-root/u-root/tree/master/integration#running-tests

My first thought would be - probably something up with the memory map. You have all the right debug options in the cmdline to get early debug output, afaict.

Also try checking out u-root before the recent merge of https://github.com/u-root/u-root/pull/1847, maybe I messed something up there.

hugelgupf, 2021-03-24:
For more debugging, I'm more likely to respond quickly on slack - slack.osfw.dev to sign up

yurinnick, 2021-03-24:
I got the same results using go 1.14 and 1.15 on latest master checkout. 

I still haven't used qemu yet, but quick test in VirtualBox was fine and ESXi started installation normally. The key difference: VBox has vtconsole available and bare metal hardware we only have serial console, but no vtconsole in u-root. I will experiment with pre #1847 revision.

Important note, we are not booting ESXi *installation*, but rather ESXi *installer* from unpacked archive, so there is no disk with completed installation yet.

yurinnick, 2021-04-21:
ESXi [mboot](https://github.com/vmware/esx-boot/blob/master/mboot/mboot.c) returned following memorymap:

```
<6>MMap[0]: 0x0 - 0x8dfff len=581632, type=7, attr=0xf
<6>MMap[1]: 0x8e000 - 0x8ffff len=8192, type=0, attr=0xf
<6>MMap[2]: 0x90000 - 0x9ffff len=65536, type=7, attr=0xf
<6>MMap[3]: 0x100000 - 0x734fffff len=1933574144, type=7, attr=0xf
<6>MMap[4]: 0x73500000 - 0x735fffff len=1048576, type=4, attr=0xf
<6>MMap[5]: 0x73600000 - 0x73cfffff len=7340032, type=7, attr=0xf
<6>MMap[6]: 0x73d00000 - 0x73d56fff len=356352, type=4, attr=0xf
<6>MMap[7]: 0x73d57000 - 0x73d57fff len=4096, type=3, attr=0xf
<6>MMap[8]: 0x73d58000 - 0x7b0cbfff len=121061376, type=4, attr=0xf
<6>MMap[9]: 0x7b0cc000 - 0x8adc2fff len=265252864, type=7, attr=0xf
<6>MMap[10]: 0x8adc3000 - 0x8adc6fff len=16384, type=6, attr=0x800000000000000f
<6>MMap[11]: 0x8adc7000 - 0x8ae0ffff len=299008, type=3, attr=0xf
<6>MMap[12]: 0x8ae10000 - 0x8b210fff len=4198400, type=4, attr=0xf
<6>MMap[13]: 0x8b211000 - 0x8b646fff len=4415488, type=3, attr=0xf
<6>MMap[14]: 0x8b647000 - 0x8b649fff len=12288, type=7, attr=0xf
<6>MMap[15]: 0x8b64a000 - 0x8c220fff len=12414976, type=3, attr=0xf
<6>MMap[16]: 0x8c221000 - 0x94421fff len=136318976, type=4, attr=0xf
<6>MMap[17]: 0x94422000 - 0x94615fff len=2048000, type=3, attr=0xf
<6>MMap[18]: 0x94616000 - 0x94725fff len=1114112, type=0, attr=0xf
<6>MMap[19]: 0x94726000 - 0x94732fff len=53248, type=6, attr=0x800000000000000f
<6>MMap[20]: 0x94733000 - 0x95732fff len=16777216, type=0, attr=0xf
<6>MMap[21]: 0x95733000 - 0x96979fff len=19165184, type=6, attr=0x800000000000000f
<6>MMap[22]: 0x9697a000 - 0x969e2fff len=430080, type=7, attr=0xf
<6>MMap[23]: 0x969e3000 - 0x96de3fff len=4198400, type=4, attr=0xf
<6>MMap[24]: 0x96de4000 - 0x96fbafff len=1929216, type=7, attr=0xf
<6>MMap[25]: 0x96fbb000 - 0x97049fff len=585728, type=4, attr=0xf
<6>MMap[26]: 0x9704a000 - 0x97065fff len=114688, type=7, attr=0xf
<6>MMap[27]: 0x97066000 - 0x97076fff len=69632, type=4, attr=0xf
<6>MMap[28]: 0x97077000 - 0x9707ffff len=36864, type=7, attr=0xf
<6>MMap[29]: 0x97080000 - 0x975f2fff len=5713920, type=4, attr=0xf
<6>MMap[30]: 0x975f3000 - 0x975f3fff len=4096, type=7, attr=0xf
<6>MMap[31]: 0x975f4000 - 0x975f5fff len=8192, type=4, attr=0xf
<6>MMap[32]: 0x975f6000 - 0x975f6fff len=4096, type=7, attr=0xf
<6>MMap[33]: 0x975f7000 - 0x975f7fff len=4096, type=4, attr=0xf
<6>MMap[34]: 0x975f8000 - 0x975f8fff len=4096, type=7, attr=0xf
<6>MMap[35]: 0x975f9000 - 0x97742fff len=1351680, type=4, attr=0xf
<6>MMap[36]: 0x97743000 - 0x97745fff len=12288, type=7, attr=0xf
<6>MMap[37]: 0x97746000 - 0x97747fff len=8192, type=4, attr=0xf
<6>MMap[38]: 0x97748000 - 0x97749fff len=8192, type=7, attr=0xf
<6>MMap[39]: 0x9774a000 - 0x9774afff len=4096, type=4, attr=0xf
<6>MMap[40]: 0x9774b000 - 0x97757fff len=53248, type=7, attr=0xf
<6>MMap[41]: 0x97758000 - 0x97759fff len=8192, type=4, attr=0xf
<6>MMap[42]: 0x9775a000 - 0x9777cfff len=143360, type=7, attr=0xf
<6>MMap[43]: 0x9777d000 - 0x97780fff len=16384, type=4, attr=0xf
<6>MMap[44]: 0x97781000 - 0x97783fff len=12288, type=7, attr=0xf
<6>MMap[45]: 0x97784000 - 0x97784fff len=4096, type=4, attr=0xf
<6>MMap[46]: 0x97785000 - 0x97786fff len=8192, type=7, attr=0xf
<6>MMap[47]: 0x97787000 - 0x97787fff len=4096, type=4, attr=0xf
<6>MMap[48]: 0x97788000 - 0x97788fff len=4096, type=7, attr=0xf
<6>MMap[49]: 0x97789000 - 0x97789fff len=4096, type=4, attr=0xf
<6>MMap[50]: 0x9778a000 - 0x9778bfff len=8192, type=7, attr=0xf
<6>MMap[51]: 0x9778c000 - 0x9778dfff len=8192, type=4, attr=0xf
<6>MMap[52]: 0x9778e000 - 0x97be9fff len=4571136, type=7, attr=0xf
<6>MMap[53]: 0x97bea000 - 0x9e979fff len=114884608, type=4, attr=0xf
<6>MMap[54]: 0x9e97a000 - 0x9ed76fff len=4182016, type=7, attr=0xf
<6>MMap[55]: 0x9ed77000 - 0x9ed79fff len=12288, type=2, attr=0xf
<6>MMap[56]: 0x9ed7a000 - 0x9ef95fff len=2211840, type=7, attr=0xf
<6>MMap[57]: 0x9ef96000 - 0x9f179fff len=1982464, type=1, attr=0xf
<6>MMap[58]: 0x9f17a000 - 0xa0179fff len=16777216, type=3, attr=0xf
<6>MMap[59]: 0xa017a000 - 0xa0579fff len=4194304, type=6, attr=0x800000000000000f
<6>MMap[60]: 0xa057a000 - 0xa0d79fff len=8388608, type=5, attr=0x800000000000000f
<6>MMap[61]: 0xa0d7a000 - 0xa0e79fff len=1048576, type=0, attr=0xf
<6>MMap[62]: 0xa0e7a000 - 0xa3279fff len=37748736, type=10, attr=0xf
<6>MMap[63]: 0xa327a000 - 0xa3479fff len=2097152, type=9, attr=0xf
<6>MMap[64]: 0xa347a000 - 0xa347afff len=4096, type=4, attr=0xf
<6>MMap[65]: 0xa347b000 - 0xa34fbfff len=528384, type=6, attr=0x800000000000000f
<6>MMap[66]: 0xa34fc000 - 0xafffffff len=212877312, type=7, attr=0xf
<6>MMap[67]: 0x100000000 - 0xc03fffefff len=821412491264, type=7, attr=0xf
<6>MMap[68]: 0xf0000 - 0xfffff len=65536, type=0, attr=0x1001
<6>MMap[69]: 0xb0000000 - 0xbfffffff len=268435456, type=0, attr=0x8
<6>MMap[70]: 0xc0000000 - 0xcfffffff len=268435456, type=11, attr=0x8000000000000001
<6>MMap[71]: 0xfe000000 - 0xfe00ffff len=65536, type=11, attr=0x8000000000000001
<6>MMap[72]: 0xc03ffff000 - 0xc03fffffff len=4096, type=0, attr=0x0
```

thebsdbox, 2021-08-10:
Also appear to be hitting this same issue:

```
~/dev# esxiboot -d /dev/nvme0n1
[  102.540336] FAT-fs (nvme0n1p5): utf8 is not a recommended IO charset for FAT filesystems, filesystem will be case sensitive!
[  102.543147] FAT-fs (nvme0n1p6): utf8 is not a recommended IO charset for FAT filesystems, filesystem will be case sensitive!
2021/08/10 15:55:40 Parsing multiboot header
2021/08/10 15:55:40 VideoMode flag is not supproted yet, trying to load anyway
2021/08/10 15:55:40 Found multiboot image
2021/08/10 15:55:40 Getting kernel entry point
2021/08/10 15:55:40 Kernel entry point at 0x4117b0
2021/08/10 15:55:40 Parsing ELF segments
2021/08/10 15:55:40 Parsing memory map
2021/08/10 15:55:40 Preparing multiboot info
2021/08/10 15:55:40 Memory map: [0x0, 0x58000) (len: 0x58000, size: 0x14, type: 1)
[0x58000, 0x59000) (len: 0x1000, size: 0x14, type: 2)
[0x59000, 0x9e000) (len: 0x45000, size: 0x14, type: 1)
[0x9e000, 0x100000) (len: 0x62000, size: 0x14, type: 2)
[0x100000, 0x40000000) (len: 0x3ff00000, size: 0x14, type: 1)
[0x40000000, 0x40400000) (len: 0x400000, size: 0x14, type: 2)
[0x40400000, 0x756f1000) (len: 0x352f1000, size: 0x14, type: 1)
[0x756f1000, 0x756f2000) (len: 0x1000, size: 0x14, type: 4)
[0x756f2000, 0x756f3000) (len: 0x1000, size: 0x14, type: 2)
[0x756f3000, 0x8a487000) (len: 0x14d94000, size: 0x14, type: 1)
[0x8a487000, 0x8a94e000) (len: 0x4c7000, size: 0x14, type: 2)
[0x8a94e000, 0x8a999000) (len: 0x4b000, size: 0x14, type: 3)
[0x8a999000, 0x8a9f9000) (len: 0x60000, size: 0x14, type: 4)
[0x8a9f9000, 0x8afff000) (len: 0x606000, size: 0x14, type: 2)
[0x8afff000, 0x8b000000) (len: 0x1000, size: 0x14, type: 1)
[0x8b000000, 0x90000000) (len: 0x5000000, size: 0x14, type: 2)
[0xe0000000, 0xf0000000) (len: 0x10000000, size: 0x14, type: 2)
[0xfe000000, 0xfe011000) (len: 0x11000, size: 0x14, type: 2)
[0xfec00000, 0xfec01000) (len: 0x1000, size: 0x14, type: 2)
[0xfed00000, 0xfed01000) (len: 0x1000, size: 0x14, type: 2)
[0xfee00000, 0xfee01000) (len: 0x1000, size: 0x14, type: 2)
[0xff000000, 0x100000000) (len: 0x1000000, size: 0x14, type: 2)
[0x100000000, 0x46e000000) (len: 0x36e000000, size: 0x14, type: 1)
2021/08/10 15:55:40 Added module jumpstrt.gz at [0x601000, 0x60102d)
2021/08/10 15:55:40 Added module useropts.gz at [0x602000, 0x602001)
2021/08/10 15:55:40 Added module features.gz at [0x603000, 0x603000)
2021/08/10 15:55:40 Added module k.b00 at [0x603000, 0x12689a4)
2021/08/10 15:55:40 Added module chardevs.b00 at [0x1269000, 0x1278424)
2021/08/10 15:55:40 Added module user.b00 at [0x1279000, 0x1459904)
2021/08/10 15:55:40 Added module procfs.b00 at [0x145a000, 0x145ff04)
2021/08/10 15:55:40 Added module uc_intel.b00 at [0x1460000, 0x1676c00)
2021/08/10 15:55:40 Added module uc_amd.b00 at [0x1677000, 0x1678bdc)
2021/08/10 15:55:40 Added module uc_hygon.b00 at [0x1679000, 0x167901c)
2021/08/10 15:55:40 Added module vmx.v00 at [0x167a000, 0x32ecb94)
2021/08/10 15:55:40 Added module vim.v00 at [0x32ed000, 0x438ef64)
2021/08/10 15:55:40 Added module sb.v00 at [0x438f000, 0x6f312f0)
2021/08/10 15:55:40 Added module s.v00 at [0x6f32000, 0x7ad1f1c)
2021/08/10 15:55:40 Added module ata_liba.v00 at [0x7ad2000, 0x7b1bb98)
2021/08/10 15:55:40 Added module ata_pata.v00 at [0x7b1c000, 0x7b27200)
2021/08/10 15:55:40 Added module ata_pata.v01 at [0x7b28000, 0x7b30c28)
2021/08/10 15:55:40 Added module ata_pata.v02 at [0x7b31000, 0x7b3a350)
2021/08/10 15:55:40 Added module ata_pata.v03 at [0x7b3b000, 0x7b446c0)
2021/08/10 15:55:40 Added module ata_pata.v04 at [0x7b45000, 0x7b4f550)
2021/08/10 15:55:40 Added module ata_pata.v05 at [0x7b50000, 0x7b59968)
2021/08/10 15:55:40 Added module ata_pata.v06 at [0x7b5a000, 0x7b62c98)
2021/08/10 15:55:40 Added module ata_pata.v07 at [0x7b63000, 0x7b6cd40)
2021/08/10 15:55:40 Added module block_cc.v00 at [0x7b6d000, 0x7b821d8)
2021/08/10 15:55:40 Added module bnxtnet.v00 at [0x7b83000, 0x7c0e760)
2021/08/10 15:55:40 Added module bnxtroce.v00 at [0x7c0f000, 0x7c5af88)
2021/08/10 15:55:40 Added module brcmfcoe.v00 at [0x7c5b000, 0x7ea47e0)
2021/08/10 15:55:40 Added module char_ran.v00 at [0x7ea5000, 0x7eb1e58)
2021/08/10 15:55:40 Added module ehci_ehc.v00 at [0x7eb2000, 0x7ec9fb0)
2021/08/10 15:55:40 Added module elxiscsi.v00 at [0x7eca000, 0x7f4abd8)
2021/08/10 15:55:40 Added module elxnet.v00 at [0x7f4b000, 0x7feb718)
2021/08/10 15:55:40 Added module hid_hid.v00 at [0x7fec000, 0x7ffb3e8)
2021/08/10 15:55:40 Added module i40en.v00 at [0x7ffc000, 0x8099c68)
[... 113 lines cut ...]
2021/08/10 15:55:40 Added module lsu_lsi_.v03 at [0xc3a1000, 0xc3e212c)
2021/08/10 15:55:40 Added module lsu_lsi_.v04 at [0xc3e3000, 0xc46e630)
2021/08/10 15:55:40 Added module lsu_smar.v00 at [0xc46f000, 0xc486684)
2021/08/10 15:55:40 Added module native_m.v00 at [0xc487000, 0xc5536c0)
2021/08/10 15:55:40 Added module qlnative.v00 at [0xc554000, 0xc813308)
2021/08/10 15:55:40 Added module rste.v00 at [0xc814000, 0xc8e27d0)
2021/08/10 15:55:40 Added module vmware_e.v00 at [0xc8e3000, 0xc91187c)
2021/08/10 15:55:40 Added module vsan.v00 at [0xc912000, 0xf41c858)
2021/08/10 15:55:40 Added module vsanheal.v00 at [0xf41d000, 0xfbc371c)
2021/08/10 15:55:40 Added module vsanmgmt.v00 at [0xfbc4000, 0x114ff58f)
2021/08/10 15:55:40 Added module xorg.v00 at [0x11500000, 0x11849b00)
2021/08/10 15:55:40 Added module imgdb.tgz at [0x1184a000, 0x11a16000)
2021/08/10 15:55:40 Added module state.tgz at [0x11a16000, 0x11a19800)
2021/08/10 15:55:40 Adding trampoline
[... cut ...] usbcore_.v00, vmkata.v00, vmkfcoe.v00, vmkplexe.v00, vmkusb.v00, vmw_ahci.v00, xhci_xhc.v00, elx_esx_.v00, btldr.t00, esx_dvfi.v00, esx_ui.v00, esxupdt.v00, weaselin.t00, lsu_hp_h.v00, lsu_inte.v00, lsu_lsi_.v00, lsu_lsi_.v01, lsu_lsi_.v02, lsu_lsi_.v03, lsu_lsi_.v04, lsu_smar.v00, native_m.v00, qlnative.v00, rste.v00, vmware_e.v00, vsan.v00, vsanheal.v00, vsanmgmt.v00, xorg.v00, imgdb.tgz, state.tgz
)
```

This worked fine in qemu, however with an intel NUC the machine hangs at the point above.