u-root/u-root #1977: add squashfs support

orangecms, 2021-04-06:
Currently, output formats are only cpio and directory. A handy third option would be squashfs, possibly with https://pkg.go.dev/github.com/distr1/distri/internal/squashfs

A regular user cannot create special devices in a directory, so the only option to convert cpio to squashfs is currently something like
```sh
mkdir /tmp/foo
cd /tmp/foo
sudo cpio -i < /tmp/u-root-image.cpio # this can be an issue
cd /tmp
mksquashfs /tmp/foo foo.sfs
```

rminnich, 2021-05-06:
this is a wonderful idea ...

orangecms, 2021-05-08:
Quite some details on squashfs can be found here:
https://dr-emann.github.io/squashfs/

rjoleary, 2021-05-27:
Would we use a squashfs writer in Go or execute mksquashfs?

One problem we experienced in fiano was that liblzma greatly outperforms (in terms of compression ratio) every lzma compression library we found written in Go. It seems we might run into the same problem here (sub-optimal compression of squashfs). The way we solved this in fiano is to use the Go compressor, but if the user passed in a special flag, it would switch to the system's xz command.