u-root/u-root #1981: With Kernel 5.10.3 kexec hang

Marshalys123, 2021-04-14:
Hi 
Recently I want to use linux kernel 5.10.3 with u-root, but I found that the localboot /pxeboot commands will hang. I chased it and found that kexec hangs after giving the second kernel. May I ask the current uboot kexec is not compatible with 5.10.3 Is the kernel?

hugelgupf, 2021-04-14:
where do they hang exactly? do you have logs you can share in an anonymized-enough-manner?

Marshalys123, 2021-04-14:
Example:
log:
~/# localboot -grub
[   45.117150] FAT-fs (nvme0n1p1): IO charset iso8859-1 not found
[   45.169339] FAT-fs (nvme1n1p1): IO charset iso8859-1 not found
[   45.238740] FAT-fs (sda1): IO charset iso8859-1 not found
2021/04/14 03:25:08 mounted: [MountPoint(path=/mnt/nvme0n1p2, device=/dev/nvme0n1p2, fs=ext4, flags=0x1, data=) MountPoint(path=/mnt
/nvme0n1p3, device=/dev/nvme0n1p3, fs=ext4, flags=0x1, data=) MountPoint(path=/mnt/nvme0n1p5, device=/dev/nvme0n1p5, fs=ext4, flags=
0x1, data=) MountPoint(path=/mnt/nvme0n1p6, device=/dev/nvme0n1p6, fs=ext4, flags=0x1, data=) MountPoint(path=/mnt/nvme1n1p2, device
=/dev/nvme1n1p2, fs=ext4, flags=0x1, data=)]
2021/04/14 03:25:08 Parsing /mnt/nvme0n1p2/boot/grub/grub.cfg
2021/04/14 03:25:08 Kernel seems to be on an other partitioin then the grub.cfg file
2021/04/14 03:25:08 fs-uuid: 38386abd-9144-4841-84bc-efcd1c8e75b9
2021/04/14 03:25:08 Kernel is on: nvme0n1p2
2021/04/14 03:25:08 Kernel seems to be on an other partitioin then the grub.cfg file
2021/04/14 03:25:08 fs-uuid: 38386abd-9144-4841-84bc-efcd1c8e75b9
2021/04/14 03:25:08 Kernel is on: nvme0n1p2
2021/04/14 03:25:08 Kernel seems to be on an other partitioin then the grub.cfg file
2021/04/14 03:25:08 fs-uuid: 38386abd-9144-4841-84bc-efcd1c8e75b9
2021/04/14 03:25:08 Kernel is on: nvme0n1p2
2021/04/14 03:25:08 Kernel seems to be on an other partitioin then the grub.cfg file
2021/04/14 03:25:08 fs-uuid: 38386abd-9144-4841-84bc-efcd1c8e75b9
2021/04/14 03:25:08 Kernel is on: nvme0n1p2
2021/04/14 03:25:08 Kernel seems to be on an other partitioin then the grub.cfg file
2021/04/14 03:25:08 fs-uuid: 38386abd-9144-4841-84bc-efcd1c8e75b9
2021/04/14 03:25:08 Kernel is on: nvme0n1p2
2021/04/14 03:25:08 Kernel seems to be on an other partitioin then the grub.cfg file
2021/04/14 03:25:08 fs-uuid: 38386abd-9144-4841-84bc-efcd1c8e75b9
2021/04/14 03:25:08 Kernel is on: nvme0n1p2
2021/04/14 03:25:08 Found 3 boot configs
2021/04/14 03:25:08   0: 'Debian GNU/Linux'
2021/04/14 03:25:08   1: 'Debian GNU/Linux, with Linux 5.4.56.bsk.3-amd64'
2021/04/14 03:25:08   2: 'Debian GNU/Linux, with Linux 5.4.56.bsk.3-amd64 (recovery mode)'
2021/04/14 03:25:08 Cannot open TPM: TPM device not available

I added debug to the linux kernel. On the x86 platform, It can run the following functions and then hang. machine_kexec_64.c machine_kexec()->
         /* now call it */
image->start = relocate_kernel((unsigned long)image->head,
(unsigned long)page_list,
image->start,
image->preserve_context,
sme_active());

rjoleary, 2021-05-27:
Did you have any luck with this?

u-root is known to work with the 5.10 kernel.

mfischer, 2021-06-24:
If you're on an ARM64 system by any chance you might want to pull in:
https://lore.kernel.org/linux-arm-kernel/20210429133533.1750721-2-maz@kernel.org/

rjoleary, 2021-06-26:
They mentioned this is x86 just above.

We should put together a doc with some kexec debug tips. The "kexec silent death" can be really difficult to debug.

One common mistake is to forget `console=ttyS0` in the command line. Everything might actually be working but without serial.

10000TB, 2021-11-23:
Nill(geshijian@bytedance.com) and I were on an email thread for this issue. I have an idea which I think worth trying to see where the bottleneck is (u-root kexec file load call `v.s.` or anything else)

1. build a kexec binary kexec-tools https://github.com/horms/kexec-tools
2. add it into a u-root initramfs (pass `-files` flag to `u-root` cmd when building, see https://github.com/horms/kexec-tools)
3. Modify https://github.com/u-root/u-root/blob/master/pkg/boot/kexec/kexec_linux_implemented.go#L30-L32 to run the cmd from `.2` to kexec the new os.

Apply this variant onto a machine or 2, and do reboots as you would do on other machines where you see the issue, and see if there is any improvement

NillatBD, 2022-05-12:
Thanks David, [Marshalys123](https://github.com/Marshalys123) was also from Bytedance, 
pengsen@bytedance.com will take over this and update later.