u-root/u-root #1988: New cbmem unexpected EOF while processing LB_TAG_CBMEM_CONSOLE

johnnylinwiwynn, 2021-04-27:
With my coreboot the new cbmem would see unexpected EOF while processing LB_TAG_CBMEM_CONSOLE, after I only revert the processing of case LB_TAG_CBMEM_CONSOLE in main_linux.go before commit 455f00d it works as before, so my coreboot cbmem console data happens to triggers the issue. Provide below are the cbmem debug logs with new and old cbmem:
New:
cbmem -l -v
2021/04/26 02:41:13 newOffsetReader(&{0xc0000c82a0}, 0x0, 0x10000)
2021/04/26 02:41:13 Looking for coreboot table at 0x00000000 65536 bytes
2021/04/26 02:41:13 header is Signature ZZZZ Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000000
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000010
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000020
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000030
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000040
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000050
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000060
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000070
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000080
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000090
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x000000a0
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x000000b0
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x000000c0
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x000000d0
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x000000e0
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x000000f0
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000100
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000110
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000120
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000130
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000140
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:13 no LBIO at 0x00000150
2021/04/26 02:41:13 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x00000160
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x00000170
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x00000180
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x00000190
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x000001a0
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x000001b0
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x000001c0
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x000001d0
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x000001e0
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
2021/04/26 02:41:16 no LBIO at 0x000001f0
2021/04/26 02:41:16 header is Signature      Header Size 0x0 HeaderCSUM 0x00000000 Table Size 0x0 TableCSUM 0x00000000 TableEntries 0
[... 611 lines cut ...]
		"LB_TAG_EXTRA_VERSION": "",
		"LB_TAG_PLATFORM_BLOB_VERSION": "2.1-0.2.2.0",
		"LB_TAG_VERSION": "4.13-3295-gb1addfe-dirty"
	},
	"BootMediaParams": {
		"Tag": 48,
		"Size": 40,
		"FMAPOffset": 50479104,
		"CBFSOffset": 50481152,
		"CBFSSize": 16627712,
		"BootMediaSize": 67108864
	},
	"VersionTimeStamp": 38,
	"Unknown": [
		44,
		54,
		55,
		64,
		50
	],
	"Ignored": [
		"LB_TAG_SPI_FLASH",
		"LB_TAG_ACPI_GNVS"
	]
}

rminnich, 2021-05-06:
oops. Is there some way you can share a dump of just the cbmem area of memory with me. Then I am going to write a test mode.

johnnylinwiwynn, 2021-05-07:
I run the coreboot/util/cbmem -x > cbmem-hexdump.txt
I modified the cbmem to print the whole memory to cbmem-hexdump2.txt
and run u-root cbmem -json > cbmem-json.txt
I share the files to
https://drive.google.com/file/d/1zja6-_Ee6wURKus5C_wzOuhDgxkoWsyb/view
Hope this helps.
Thanks!

PR #2004 (2021-05-12): cmds/exp/cbmem: add test and correctly handle wrapping
The cbmem console can wrap and start writing new console
logs at the start. Correctly handle this case.

Add a test for a bogus cursor.

Add checks for bad cbmem pointers and handle them if they
occur.
fixes #1988
commit d562d9ea (2021-05-12): cmds/exp/cbmem: add test and correctly handle wrapping

The cbmem console can wrap and start writing new console
logs at the start. Correctly handle this case.

Add a test for a bogus cursor.

Add checks for bad cbmem pointers and handle them if they
occur.
fixes #1988
closed 2021-05-12 by rminnich