u-root/u-root #2008: u-root can not include a command from another go tree (it used to) [is the answer to just use gobusybox?]

rminnich, 2021-05-17:
rminnich@t510:~/go/src/github.com/u-root/u-root$ u-root core github.com/u-root/webboot/cmds/wifi
2021/05/17 02:54:25 Disabling CGO for u-root...
2021/05/17 02:54:25 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib/go-1.14 GOPATH=/home/rminnich/go CGO_ENABLED=0 GO111MODULE=off
2021/05/17 02:54:25 Skipping package "/home/rminnich/go/src/github.com/u-root/u-root/cmds/core/bind": no buildable Go source files in /home/rminnich/go/src/github.com/u-root/u-root/cmds/core/bind
2021/05/17 02:54:56 Successfully built "/tmp/initramfs.linux_amd64.cpio" (size 15940896).
rminnich@t510:~/go/src/github.com/u-root/u-root$ sudo ~/go/bin/testramfs -i /tmp/initramfs.linux_amd64.cpio 
2021/05/17 02:56:37 Welcome to u-root!
                              _
   _   _      _ __ ___   ___ | |_
  | | | |____| '__/ _ \ / _ \| __|
  | |_| |____| | | (_) | (_) | |_
   \__,_|    |_|  \___/ \___/ \__|

init: 2021/05/17 02:56:37 no modules found matching '/lib/modules/*.ko'
(unreachable)/home/rminnich/go/src/github.com/u-root/u-root# wifi
2021/05/17 02:56:39 Invalid busybox command: ["/bbin/wifi"]
Exception: wifi exited with 1
[tty], line 1: wifi
(unreachable)/home/rminnich/go/src/github.com/u-root/u-root# ls -l /bbin/wifi
Lrwxrwxrwx root 0 2 May 17 02:56 wifi -> bb
(unreachable)/home/rminnich/go/src/github.com/u-root/u-root# bb
wifi:	"/bbin/wifi"	NO COMMAND
There is at least one problem. Known causes:
At least two initrds -- one compiled in to the kernel, a second supplied by the bootloader.
The initrd cpio was changed after creation or merged with another one.
When the initrd was created, files were inserted into /bbin by mistake.
Post boot, files were added to /bbin.
(unreachable)/home/rminnich/go/src/github.com/u-root/u-root#      

there is a symlink for wifi, but there is "no command"

we need a better way to tell what u-root is doing nowadays I guess.

This one just flat out fails, and it used to work just fine:
rminnich@t510:~/go/src/github.com/u-root/u-root$ go get github.com/nsf/godit
go: downloading github.com/nsf/godit v0.0.0-20190302080138-f088c9f5e84a
go: github.com/nsf/godit upgrade => v0.0.0-20190302080138-f088c9f5e84a
go: finding module for package github.com/nsf/tulib
go: finding module for package github.com/nsf/termbox-go
go: downloading github.com/nsf/termbox-go v1.1.1
go: downloading github.com/nsf/tulib v0.0.0-20140820125801-b257bdd8b41a
go: found github.com/nsf/termbox-go in github.com/nsf/termbox-go v1.1.1
go: found github.com/nsf/tulib in github.com/nsf/tulib v0.0.0-20140820125801-b257bdd8b41a
go: downloading github.com/mattn/go-runewidth v0.0.9
rminnich@t510:~/go/src/github.com/u-root/u-root$ u-root core github.com/nsf/godit
2021/05/17 02:59:06 Disabling CGO for u-root...
2021/05/17 02:59:06 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/lib/go-1.14 GOPATH=/home/rminnich/go CGO_ENABLED=0 GO111MODULE=off
2021/05/17 02:59:06 Skipping package "/home/rminnich/go/src/github.com/u-root/u-root/cmds/core/bind": no buildable Go source files in /home/rminnich/go/src/github.com/u-root/u-root/cmds/core/bind
2021/05/17 02:59:12 error building: cannot find package "github.com/nsf/godit" in any of:
	/usr/lib/go-1.14/src/github.com/nsf/godit (from $GOROOT)
	/home/rminnich/go/src/github.com/nsf/godit (from $GOPATH)
rminnich@t510:~/go/src/github.com/u-root/u-root$ go install github.com/nsf/godit
rminnich@t510:~/go/src/github.com/u-root/u-root$ godit
rminnich@t510:~/go/src/github.com/u-root/u-root$

hugelgupf, 2021-05-17:
> Assigned #2008 <https://github.com/u-root/u-root/issues/2008> to
> @hugelgupf <https://github.com/hugelgupf>.

I didn't read through all of this, but...

go get does not git clone + download a repo into GOPATH anymore. So it's
actually truly not there.

GO111MODULE=off go get <repo>

will do it.

rminnich, 2021-05-18:
but this does not fix the other problems ... sadly.

rjoleary, 2021-05-27:
It looks like it's still under GOPATH, just GOPATH/pkg/mod/github.com/u-root/u-root instead of GOPATH/src/github.com/u-root/u-root

rminnich, 2021-06-03:
gobusybox is the answer to all my needs here, so I'm closing this :-)

closed 2024-02-20 by hugelgupf