u-root/u-root #2033: Automatically regenerate man command from the u-root command

rjoleary, 2021-06-07:
This would have two benefits:

* This would avoid us from having to regenerate the command manually (see for example https://github.com/u-root/u-root/pull/2032).
* The man command would have the documentation only for the commands included in your initramfs.

JohnAZoidberg, 2021-06-15:
It would be great to also have the manpages also available outside of the shell.
For example on a documentation website, like the LinuxBoot book.
Perhaps we can have a simple script to turn the extracted manuals to Markdown or first to roff and then HTML/markdown.

I'm trying to figure out how to run the generation script but I can't figure it out: https://github.com/u-root/u-root/pull/2032#issuecomment-861128388

rjoleary, 2021-06-15:
You can see the list of commands and html documentation here: https://pkg.go.dev/github.com/u-root/u-root@master#section-directories (the formatting used to be nicer on godoc.org)