u-root/u-root #2046: u-root .tk domain name

mricon, 2021-06-14:
Looks like u-root (.) tk is showing "Owned by ALI". I'm not sure if the project still has any association with that domain name, but a lot of old documentation is still pointing at it. Perhaps you can re-establish control over the .tk domain so it's not potentially serving malicious content?

See [this reddit thread](https://www.reddit.com/r/linux/comments/nzlv8s/linux_foundation_video_description_on_youtube_may/)

rjoleary, 2021-06-19:
It appears we were afflicted by https://docs.github.com/en/pages/configuring-a-custom-domain-for-your-github-pages-site/about-custom-domains-and-github-pages#updating-custom-domains-when-your-github-pages-site-is-disabled

In https://github.com/u-root/u-root/commit/0f27350795cce9bcb98642b0ebdcdea7a5f143c6#diff-00d22ea612a748d8ddd4e5bf6a203591ae83569e7b16f50a11fa6d31a5bedddd, the tk domain was removed from the CNAME file. The DNS record still points to github, but unassigned to a project. I found a project which added a CNAME file to their project to claim our domain. Beside u-root, there are a number of other sites now "Owned by ALI" -- I assume it is some form of unfriendly bot doing this automatically.

I will see if I can resolve this with github support.

rjoleary, 2021-07-07:
u-root.tk now redirects to u-root.org.

closed 2021-07-07 by rjoleary