u-root/u-root #2051: Implement parsing of GRUB2 submenus

JohnAZoidberg, 2021-06-17:
Update: I initially thought there's a bug in u-root that duplicates the menu entries but it's actually the case that there are many submenus with the same entries but different grub parameters: https://github.com/NixOS/nixpkgs/blob/master/nixos/modules/installer/cd-dvd/iso-image.nix#L334
I'll try to understand how GRUB2 submenus work and how u-root might want to support them.

I attached the NixOS installation iso to QEMU and use the `boot` command to boot it.
The generated boot menu shows some boot entries many times.

I have the entries:

```
LABEL boot
MENU LABEL NixOS 21.05pre289339.83d907fd760 Installer
LINUX /boot/bzImage
APPEND init=/nix/store/mqsfdrqf3p7hqds3j4p7fhk3r74d9s9p-nixos-system-nixos-21.05pre289339.83d907fd760/init root=LABEL=nixos-minimal-21.05-x86_64 boot.shell_on_fail loglevel=4
INITRD /boot/initrd

# A variant to boot with 'nomodeset'
LABEL boot-nomodeset
MENU LABEL NixOS 21.05pre289339.83d907fd760 Installer (nomodeset)
LINUX /boot/bzImage
APPEND init=/nix/store/mqsfdrqf3p7hqds3j4p7fhk3r74d9s9p-nixos-system-nixos-21.05pre289339.83d907fd760/init root=LABEL=nixos-minimal-21.05-x86_64 boot.shell_on_fail loglevel=4 nomodeset
INITRD /boot/initrd

# A variant to boot with 'copytoram'
LABEL boot-copytoram
MENU LABEL NixOS 21.05pre289339.83d907fd760 Installer (copytoram)
LINUX /boot/bzImage
APPEND init=/nix/store/mqsfdrqf3p7hqds3j4p7fhk3r74d9s9p-nixos-system-nixos-21.05pre289339.83d907fd760/init root=LABEL=nixos-minimal-21.05-x86_64 boot.shell_on_fail loglevel=4 copytoram
INITRD /boot/initrd

# A variant to boot with verbose logging to the console
LABEL boot-debug
MENU LABEL NixOS 21.05pre289339.83d907fd760 Installer (debug)
LINUX /boot/bzImage
APPEND init=/nix/store/mqsfdrqf3p7hqds3j4p7fhk3r74d9s9p-nixos-system-nixos-21.05pre289339.83d907fd760/init root=LABEL=nixos-minimal-21.05-x86_64 boot.shell_on_fail loglevel=4 loglevel=7
INITRD /boot/initrd

# A variant to boot with a serial console enabled
LABEL boot-serial
MENU LABEL NixOS 21.05pre289339.83d907fd760 Installer (serial console=ttyS0,115200n8)
LINUX /boot/bzImage
APPEND init=/nix/store/mqsfdrqf3p7hqds3j4p7fhk3r74d9s9p-nixos-system-nixos-21.05pre289339.83d907fd760/init root=LABEL=nixos-minimal-21.05-x86_64 boot.shell_on_fail loglevel=4 console=ttyS0,115200n8
INITRD /boot/initrd

LABEL memtest
MENU LABEL Memtest86+
LINUX /boot/memtest.bin
APPEND
```

The following are added by u-root 10 times each:
```
NixOS 21.05pre289339.83d907fd760 Installer
NixOS 21.05pre289339.83d907fd760 Installer (nomodeset)
NixOS 21.05pre289339.83d907fd760 Installer (copytoram)
NixOS 21.05pre289339.83d907fd760 Installer (debug)```
```

but those two are added only once:

```
NixOS 21.05pre289339.83d907fd760 Installer (serial console=ttyS0,115200n8)
Memtest86+
```

```
~/# boot
2021/06/17 07:48:24 Booting from the following block devices: [BlockDevice(name=vda, fs_uuid=455d7919-56a4-45b0-8c47-51955d53a599) BlockDevice(name=vdb, fs_uuid=) BlockDevice(name=vdb1, fs_uuid=fabe-1afd) BlockDevice(name=vdc, fs_uuid=) BlockDevice(name=vdc1, fs_uuid=) BlockDevice(name=vdc2, fs_uuid=1234-5678)]
ISOFS: Unable to identify CD-ROM format.
ISOFS: Unable to identify CD-ROM format.
ISOFS: Unable to identify CD-ROM format.
[... 168 lines cut ...]
```

**To Reproduce**
```
qemu-system-x86_64 \
	-L . \
	-net none \
	-enable-kvm \
	-nographic \
	-m 1024M \
	-drive file=nixos-21.05pre289339.83d907fd760-x86_64-linux.iso,format=raw,if=virtio,media=cdrom,id=hd2 \
	-kernel bzImage \
	-initrd initramfs.linux_amd64.cpio
```

where

- `nixos-21.05pre289339.83d907fd760-x86_64-linux.iso` is downloaded from https://channels.nixos.org/nixos-21.05/latest-nixos-minimal-x86_64-linux.iso
- `initramfs.linux_amd64.cpio` is built by `u-root core boot`

**Expected behavior**
I expect each grub entry to show up in the u-root boot menu only once.

**Additional context**
Add any other context about the problem here.

JohnAZoidberg, 2021-06-17:
Apparently one other duplication of the entries is because there's grub and isolinux configuration.
Perhaps u-root should highlight where the entries were extracted from.

JohnAZoidberg, 2021-06-17:
Maybe also make it possible to view the config (kernel, initrd, cmdline) of each option before executing it, to verify it's the right one. Currently, it seems, you have to tell by the name.

walterchris, 2022-01-21:
Does this relate to #2237 ?

johnnylinwiwynn, 2022-01-22:
> Does this relate to #2237 ?

https://github.com/u-root/u-root/pull/2237 only focuses on improving GRUB BLSCFG support, the duplicate boot menu entries is still there.