u-root/u-root #2055: Proposal: cmds/contrib directory

rjoleary, 2021-06-26:
The "cmds/contrib" directory would be for commands specific to a particular organization or use-case, not part of core.

For example, the following commands would be moved to "cmds/contrib": disk_unlock, fixrsdp, smbios_transfer, smn, srvfiles, lockmsrs

They are commands which are reasonably polished and no longer 'exp'erimental, but are not a good fit for being considered part of the core set of commands.

Many other projects have the concept of a "contrib" folder (see https://softwareengineering.stackexchange.com/questions/252053/whats-in-the-contrib-folder/339925), "The contrib directory is used as an unorganized (or, at best, lightly organized) bin of various useful things contributed by the community around the software, but which is not necessarily a good candidate for being a proper part of the software."

hugelgupf, 2021-06-26:
I like the idea

rjoleary, 2021-07-24:
It's created.

closed 2021-07-24 by rjoleary