u-root/u-root #2059: Support autoloading kernel modules based on sysfs

twitchyliquid64, 2021-06-28:
**Is your feature request related to a problem? Please describe.**

Its not easy to get the right kernel modules to load, currently you need to:
1. load them manually or compile them into the kernel image
2. bundle udev (ew)
3. do some unholy bash to parse `modalias` and attempt to load based on that, which I don't think works for u-root as `modprobe` i dont think can load by alias

**Describe the solution you'd like**
It would be nice to have a command in u-root that can do basic module loading. Something like:

1. Scan a few subsystems in sysfs (pci, scsi, virtio etc)
2. Read and parse the `modalias` for devices
3. Read the alias file in `/lib/modules/blah` to map to a module, and load it.

Happy to write an initial PR if that would be something we could get merged.

**Describe alternatives you've considered**
Some unholy eldricht machination in bash to do the same.

**Additional context**
My favorite reference on modalias is https://wiki.archlinux.org/title/Modalias. Otherwise ive just been browsing kernel source.

jlowellwofford, 2021-06-28:
There are some existing tools like this that you can build into your u-root build. I currently use https://github.com/bensallen/modscan

twitchyliquid64, 2021-06-28:
Oooh awesome!

Any chance we could get this merged into the tree? or am I being too pedantic about where things are located?

I also noticed it does alias matching based entirely on string wildcards - is that what udev does too? I couldnt find a definite answer about whether it was better to parse all the different modalias representations and match manually, or whether simple string matching was the gold standard.

yin-zt, 2025-01-21:
> There are some existing tools like this that you can build into your u-root build. I currently use https://github.com/bensallen/modscan

i really appreciate your suggestion