u-root/u-root #2062: go test ... fail because of go modules mess

10000TB, 2021-07-08:
**Describe the bug**
go modules messed up

`go test ...` does not run because of module issues 

**To Reproduce**
Steps to reproduce the behavior: from HEAD, with golang `1.15.2`, and run `go test ...`

**Expected behavior**
tests go through

**Additional context**
n/a

10000TB, 2021-07-08:
e.g. 

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root$ go test ...
go: finding module for package golang.org/x/image/font/inconsolata
go: finding module for package golang.org/x/mobile/gl
go: finding module for package github.com/hashicorp/consul/api
go: finding module for package github.com/gorilla/mux
go: finding module for package golang.org/x/mobile/event/lifecycle
go: finding module for package github.com/coreos/pkg/capnslog
go: finding module for package golang.org/x/mobile/geom
go: finding module for package github.com/lyft/protoc-gen-star/lang/go
go: finding module for package github.com/pborman/uuid
go: finding module for package github.com/samuel/go-zookeeper/zk
go: finding module for package github.com/soheilhy/cmux
go: finding module for package github.com/openzipkin/zipkin-go
go: finding module for package github.com/influxdata/influxdb/client/v2
go: finding module for package google.golang.org/api/transport/grpc
go: finding module for package github.com/lightstep/lightstep-tracer-go
go: finding module for package golang.org/x/time/rate
go: finding module for package go.opencensus.io/trace
go: finding module for package github.com/sony/gobreaker
go: finding module for package golang.org/x/image/font
go: finding module for package google.golang.org/api/bigquery/v2
go: finding module for package google.golang.org/api/transport/http
go: finding module for package github.com/coreos/bbolt
go: finding module for package github.com/urfave/cli
go: finding module for package github.com/lyft/protoc-gen-star
go: finding module for package google.golang.org/api/googleapi
go: finding module for package google.golang.org/api/support/bundler
go: finding module for package golang.org/x/mobile/event/key
go: finding module for package github.com/dgrijalva/jwt-go
go: finding module for package github.com/neelance/astrewrite
go: finding module for package google.golang.org/api/storage/v1
go: finding module for package github.com/google/btree
go: finding module for package github.com/xiang90/probing
go: finding module for package github.com/coreos/go-systemd/daemon
go: finding module for package github.com/golang/groupcache/lru
go: finding module for package google.golang.org/api/clouddebugger/v2
go: finding module for package github.com/envoyproxy/protoc-gen-validate/tests/harness/cases/gogo
go: finding module for package golang.org/x/mobile/event/size
go: finding module for package github.com/BurntSushi/xgb/xproto
go: finding module for package github.com/coreos/go-systemd/util
go: finding module for package github.com/afex/hystrix-go/hystrix
go: finding module for package go.opencensus.io/tag
go: finding module for package github.com/performancecopilot/speed
go: finding module for package github.com/ghodss/yaml
go: finding module for package github.com/BurntSushi/xgb/render
go: finding module for package github.com/aws/aws-sdk-go-v2/aws
go: finding module for package github.com/streadway/handy/breaker
go: finding module for package github.com/oklog/oklog/pkg/group
go: finding module for package github.com/openzipkin/zipkin-go/reporter/http
go: finding module for package go.uber.org/zap
go: finding module for package google.golang.org/grpc/transport
go: finding module for package github.com/shurcooL/httpfs/vfsutil
go: finding module for package github.com/grpc-ecosystem/grpc-gateway/runtime
go: finding module for package github.com/olekukonko/tablewriter
go: finding module for package github.com/google/martian/martianhttp
go: finding module for package github.com/nats-io/gnatsd/server
go: finding module for package google.golang.org/api/transport
go: finding module for package github.com/BurntSushi/xgb/shm
go: finding module for package github.com/google/pprof/profile
go: finding module for package github.com/streadway/amqp
go: finding module for package go.opencensus.io/stats/view
go: finding module for package github.com/openzipkin-contrib/zipkin-go-opentracing
go: finding module for package sourcegraph.com/sourcegraph/appdash/opentracing
go: finding module for package github.com/envoyproxy/protoc-gen-validate/tests/harness/gogo
go: finding module for package go.opencensus.io/plugin/ochttp
go: finding module for package google.golang.org/api/container/v1
go: finding module for package golang.org/x/image/math/fixed
[... 181 lines cut ...]
go: found golang.org/x/image/font/inconsolata in golang.org/x/image v0.0.0-20210628002857-a66eb6448b8d
go: found google.golang.org/api/iterator/testing in google.golang.org/api v0.50.0
go: found github.com/casbin/casbin/persist/file-adapter in github.com/casbin/casbin v1.9.1
go: found github.com/pact-foundation/pact-go/dsl in github.com/pact-foundation/pact-go v1.5.3
go: found go.opencensus.io/plugin/ochttp/propagation/tracecontext in go.opencensus.io v0.23.0
go: found github.com/openzipkin/zipkin-go/reporter/recorder in github.com/openzipkin/zipkin-go v0.2.5
go: found github.com/nats-io/gnatsd/server in github.com/nats-io/gnatsd v1.4.1
go: found github.com/shurcooL/go/importgraphutil in github.com/shurcooL/go v0.0.0-20200502201357-93f07166e636
go: found github.com/google/gofuzz in github.com/google/gofuzz v1.2.0
go: github.com/coreos/etcd/etcdctl/ctlv2/command imports
	github.com/coreos/bbolt: github.com/coreos/bbolt@v1.3.6: parsing go.mod:
	module declares its path as: go.etcd.io/bbolt
	        but was required as: github.com/coreos/bbolt
```

One can do a manual edit to `go.mod` to work around this issue, by `replace github.com/coreos/bbolt => go.etcd.io/bbolt v1.3.3`. See https://github.com/etcd-io/etcd/issues/11739#issue-591769125. But that only fixes one issue among others in the same category.

It then complains similar issue for other module, e.g. 

```
go: github.com/go-kit/kit/examples/stringsvc4 imports
	github.com/nats-io/go-nats: github.com/nats-io/go-nats@v1.8.1: parsing go.mod:
	module declares its path as: github.com/nats-io/nats.go
	        but was required as: github.com/nats-io/go-nats
```

10000TB, 2021-07-08:
It looks like setting `GOPROXY=direct` would workaround the issue. See https://github.com/go-kit/kit/issues/940#issuecomment-570984487

hugelgupf, 2021-07-08:
Where is that even used?

10000TB, 2021-07-08:
tried it on both `go fmt ` and `go test ...`

Looks like there is no issue in the end

```
go: found github.com/crossdock/crossdock-go in github.com/crossdock/crossdock-go v0.0.0-20160816171116-049aabb0122b
go: go.etcd.io/etcd/auth imports
	github.com/coreos/etcd/Godeps/_workspace/src/github.com/coreos/pkg/capnslog: package provided by github.com/coreos/etcd at latest version v2.3.8+incompatible but not at required version v3.3.10+incompatible
```

10000TB, 2021-07-08:
`GO111MODULE=off`

10000TB, 2021-07-08:
`GO111MODULE=off go test ./...` seems to work

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root$ GO111MODULE=off go test ./...

ok  	github.com/u-root/u-root	249.505s
?   	github.com/u-root/u-root/cmds/boot/boot	[no test files]
?   	github.com/u-root/u-root/cmds/boot/fbnetboot	[no test files]
?   	github.com/u-root/u-root/cmds/boot/fitboot	[no test files]
?   	github.com/u-root/u-root/cmds/boot/localboot	[no test files]
?   	github.com/u-root/u-root/cmds/boot/pxeboot	[no test files]
?   	github.com/u-root/u-root/cmds/boot/systemboot	[no test files]
ok  	github.com/u-root/u-root/cmds/core/basename	(cached)
?   	github.com/u-root/u-root/cmds/core/blkid	[no test files]
ok  	github.com/u-root/u-root/cmds/core/cat	(cached)
ok  	github.com/u-root/u-root/cmds/core/chmod	(cached)
ok  	github.com/u-root/u-root/cmds/core/chroot	(cached)
ok  	github.com/u-root/u-root/cmds/core/cmp	(cached)
ok  	github.com/u-root/u-root/cmds/core/comm	(cached)
ok  	github.com/u-root/u-root/cmds/core/cp	(cached)
ok  	github.com/u-root/u-root/cmds/core/cpio	(cached)
ok  	github.com/u-root/u-root/cmds/core/date	(cached)
ok  	github.com/u-root/u-root/cmds/core/dd	(cached)
?   	github.com/u-root/u-root/cmds/core/df	[no test files]
ok  	github.com/u-root/u-root/cmds/core/dhclient	(cached)
ok  	github.com/u-root/u-root/cmds/core/dirname	(cached)
ok  	github.com/u-root/u-root/cmds/core/dmesg	(cached)
ok  	github.com/u-root/u-root/cmds/core/echo	(cached)
?   	github.com/u-root/u-root/cmds/core/elvish	[no test files]
?   	github.com/u-root/u-root/cmds/core/elvish/build	[no test files]
ok  	github.com/u-root/u-root/cmds/core/elvish/edit	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/completion	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/eddefs	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/highlight	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/history	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/lastcmd	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/location	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/prompt	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/tty	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/edit/ui	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/eval	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/eval/bundled	(cached)
?   	github.com/u-root/u-root/cmds/core/elvish/eval/store	[no test files]
ok  	github.com/u-root/u-root/cmds/core/elvish/eval/vals	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/eval/vars	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/getopt	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/glob	(cached)
?   	github.com/u-root/u-root/cmds/core/elvish/hash	[no test files]
?   	github.com/u-root/u-root/cmds/core/elvish/hashmap	[no test files]
ok  	github.com/u-root/u-root/cmds/core/elvish/parse	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/parse/parseutil	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/program	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/program/shell	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/runtime	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/store	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/store/storedefs	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/sys	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/tt	(cached)
ok  	github.com/u-root/u-root/cmds/core/elvish/util	0.016s
?   	github.com/u-root/u-root/cmds/core/elvish/vector	[no test files]
ok  	github.com/u-root/u-root/cmds/core/false	(cached)
?   	github.com/u-root/u-root/cmds/core/find	[no test files]
ok  	github.com/u-root/u-root/cmds/core/free	(cached)
ok  	github.com/u-root/u-root/cmds/core/fusermount	(cached)
?   	github.com/u-root/u-root/cmds/core/gpgv	[no test files]
?   	github.com/u-root/u-root/cmds/core/gpt	[no test files]
ok  	github.com/u-root/u-root/cmds/core/grep	(cached)
?   	github.com/u-root/u-root/cmds/core/gzip	[no test files]
ok  	github.com/u-root/u-root/cmds/core/hexdump	(cached)
?   	github.com/u-root/u-root/cmds/core/hostname	[no test files]
[... 281 lines cut ...]
ok  	github.com/u-root/u-root/pkg/uroot/initramfs	(cached)
ok  	github.com/u-root/u-root/pkg/uroot/initramfs/test	(cached) [no tests to run]
ok  	github.com/u-root/u-root/pkg/uroot/test/bar	(cached) [no tests to run]
?   	github.com/u-root/u-root/pkg/uroot/test/foo	[no test files]
?   	github.com/u-root/u-root/pkg/uroot/test/gopath1/src/foo	[no test files]
?   	github.com/u-root/u-root/pkg/uroot/test/gopath1/src/os	[no test files]
?   	github.com/u-root/u-root/pkg/uroot/test/gopath2/src/mypkga	[no test files]
?   	github.com/u-root/u-root/pkg/uroot/test/gopath2/src/mypkgb	[no test files]
ok  	github.com/u-root/u-root/pkg/uroot/util	(cached) [no tests to run]
ok  	github.com/u-root/u-root/pkg/uzip	(cached)
ok  	github.com/u-root/u-root/pkg/vfile	(cached)
ok  	github.com/u-root/u-root/pkg/vmtest	(cached) [no tests to run]
ok  	github.com/u-root/u-root/pkg/vmtest/internal/json2test	(cached) [no tests to run]
ok  	github.com/u-root/u-root/pkg/vpd	(cached)
ok  	github.com/u-root/u-root/pkg/watchdog	(cached) [no tests to run]
ok  	github.com/u-root/u-root/pkg/watchdogd	(cached) [no tests to run]
?   	github.com/u-root/u-root/tools/build_perf	[no test files]
?   	github.com/u-root/u-root/tools/checklicenses	[no test files]
?   	github.com/u-root/u-root/tools/makebb	[no test files]
?   	github.com/u-root/u-root/tools/makebbmain	[no test files]
?   	github.com/u-root/u-root/tools/mkinitramfs	[no test files]
?   	github.com/u-root/u-root/tools/testramfs	[no test files]
ok  	github.com/u-root/u-root/tools/vpdbootmanager	(cached)
FAIL
```

10000TB, 2021-07-09:
`GO111MODULE=off go test ./... ` being working consistently. It looks like CI also pass reliably

might be my local setup

closed 2021-07-09 by 10000TB