u-root/u-root #2063: Print a nice "out of memory" log message instead of OOM for pxeboot

rjoleary, 2021-07-12:
If the initramfs cannot fit in RAM, the bootloader will inevitably OOM.

Instead, it would be nicer if the relevant code returned a useful `error` which is logged such as "out of memory" and the bootloader can continue running gracefully.

One possibility is to:

* During the download, listen on mem_notify https://lwn.net/Articles/267013/
* If the system becomes low on memory, cancel the download, free the buffer and return an error message.

10000TB, 2021-07-19:
If it could not get the full NetBoot payload, is it still worth proceeding ?

If we do as suggested, that is LBYL, look before you leap. Current is (sort of, not exactly) EAFP. From my view, either way, we fail the secure binary, and Golang panic (with verbose trace) or OOM killer trace is verbose enough. If we are not gonna execute hooks right before it would OOM, it does not seems worth the effort.

rjoleary, 2021-07-27:
In webboot, the experience is a lot better. For example in https://github.com/u-root/webboot/issues/247. It prints the error "write /tmp/iso-download-758024287: no space left on device" and the program continues running rather than an OOM. I wonder how the mechanism differs here.

fotonick, 2021-07-27:
I'd say the relevant principle here is more "boots, not bricks", a la Ron.