u-root/u-root #2071: trampoline code needs update for Go 1.17+

thanm, 2021-07-22:
Looking at the this code in uroot:

https://github.com/u-root/u-root/blob/53b76439a21d5f7b808cfb096f443d04575c4fbf/pkg/boot/multiboot/internal/trampoline/trampoline_linux_amd64.go#L34

The scheme used to collect the starting PC of the various assembly routines is not going to work properly with Go 1.17 (soon to be released). 

With 1.17 the go team is rolling out a new register ABI, and as a result when you refer to an assembly function from Go you'll get a wrapper stub instead of the actual function. More details at

https://go.googlesource.com/proposal/+/master/design/27539-internal-abi.md#proposal

What this means for u-root is that the code cited above needs to be rewritten. 

I will see if I can send a pull request shortly.

thanm, 2021-07-22:
Sent a PR that should fix this: https://github.com/thanm/u-root/tree/regabi_fix_for_issue_2071

I don't have a lot of experience with github PR workflow, so I will probably be making mistakes along the way, let me know what I am doing wrong. Thanks.

thanm, 2021-07-22:
@hugelgupf

rjoleary, 2021-09-07:
I'm assuming this was fixed in https://github.com/u-root/u-root/pull/2072

closed 2021-09-07 by rjoleary