u-root/u-root #2079: Add "PXE UROOT" to DHCPv6 vendor class

rjoleary, 2021-07-30:
**Is your feature request related to a problem? Please describe.**
This is extra diagnosis information to compare u-root DHCP packets from standard DHCP packets. It is already this way for DHCPv4.

**Describe the solution you'd like**
`dhcpv6.WithOption(dhcpv6.ParseOptVendorClass([]byte("PXE UROOT")))`

**Describe alternatives you've considered**
We can also include BIOS version information from the SMBIOS tables. Perhaps append the env variable `$UROOT_VENDOR_CLASS` to "PXE_UROOT" so this is configurable.

hugelgupf, 2021-07-30:
Don't use Parse. That is not a validly formated byte-vendor-opts either.
It's missing the enterprise number.

Just create that object as &dhcpv6.OptVendorClass{EnterpriseNumber: N,
Data: [][]byte{[]byte("PXE UROOT")}}

rjoleary, 2021-07-31:
What would be used as the enterprise number? Is there one already for LinuxBoot or Linux foundation?