u-root/u-root #2082: uefivars is still using the deprecated /sys/firmware/efi/vars interface

rgl, 2021-07-31:
**Describe the bug**

The [uefivars package](https://github.com/u-root/u-root/tree/master/pkg/uefivars) is still using the deprecated `/sys/firmware/efi/vars` interface.

**To Reproduce**

Run u-root under linux 5.4.x compiled with `CONFIG_EFI_VARS=n` and `CONFIG_EFIVAR_FS=y`.

**Expected behavior**

I expected the [cmds/exp/bootvars](https://github.com/u-root/u-root/tree/master/cmds/exp/bootvars) (which is using `/sys/firmware/efi/vars` at https://github.com/u-root/u-root/blob/b81eccb0c419d45255b534c0b3644a191748adf0/pkg/uefivars/vars.go#L24-L25) to work out-of-the-box by using the `/sys/firmware/efi/efivars` interface.

For more information about the two interfaces see [Linux kernel config options for UEFI](https://wiki.archlinux.org/title/Unified_Extensible_Firmware_Interface#Linux_kernel_config_options_for_UEFI).

MDr164, 2021-09-06:
I'll be working on that now

rgl, 2023-10-15:
@MDr164 did you got a change to look at this?

MDr164, 2023-10-16:
> @MDr164 did you got a change to look at this?

We added https://github.com/u-root/u-root/blob/main/cmds/exp/efivarfs/efivarfs.go quite some time ago which uses the efivarfs, maybe that does what you're looking for?

rgl, 2023-10-16:
@MDr164 I didn't notice that new command. thank you!

Though, it seems https://github.com/u-root/u-root/blob/main/pkg/efivarfs/varfs.go#L193-L195 does not work in my machine, as it does not return anything (and `ls -la` does):

```console
/# efivarfs list                  
/# ls -la /sys/firmware/efi/efivars/
drwxr-xr-x root 0 0   Oct 16 19:28 .
-rw-r--r-- root 0 66  Oct 16 19:28 Boot0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 106 Oct 16 19:28 Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 92  Oct 16 19:28 Boot0002-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 6   Oct 16 19:28 BootCurrent-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 8   Oct 16 19:28 BootOptionSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 10  Oct 16 19:28 BootOrder-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 126 Oct 16 19:28 ConIn-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 695 Oct 16 19:28 ConInDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 77  Oct 16 19:28 ConOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 661 Oct 16 19:28 ConOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 77  Oct 16 19:28 ErrOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 661 Oct 16 19:28 ErrOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 18  Oct 16 19:28 Key0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 18  Oct 16 19:28 Key0001-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 8   Oct 16 19:28 Lang-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 17  Oct 16 19:28 LangCodes-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 8   Oct 16 19:28 MTC-eb704011-1402-11d3-8e77-00a0c969723b
-rw-r--r-- root 0 12  Oct 16 19:28 OsIndicationsSupported-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 7   Oct 16 19:28 PlatformLang-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 22  Oct 16 19:28 PlatformLangCodes-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 112 Oct 16 19:28 PlatformRecovery0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 5   Oct 16 19:28 SecureBoot-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 5   Oct 16 19:28 SetupMode-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 68  Oct 16 19:28 SignatureSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 6   Oct 16 19:28 Timeout-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 5   Oct 16 19:28 VarErrorFlag-04b37fe8-f6ae-480b-bdd5-37d98c5e89aa
-rw-r--r-- root 0 5   Oct 16 19:28 VendorKeys-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- root 0 8   Oct 16 19:28 certdb-d9bee56e-75dc-49d9-b4d7-b534210f637a
-rw-r--r-- root 0 8   Oct 16 19:28 certdbv-d9bee56e-75dc-49d9-b4d7-b534210f637a

```

MDr164, 2023-10-18:
That's odd, I guess I'll have to check if the code regressed because we have been using it in a different project and that one still worked fine last time I checked.