u-root/u-root #2198: Cannot makebb for dependency of gdamore/tcell

hagna, 2021-11-29:
This script duplicates the error message "could not import github.com/gdamore/tcell (type-checking package "github.com/gdamore/tcell" failed"
```
#!/bin/bash

mkdir bug
cd bug
mkdir -p src/simple
mkdir -p src/hello
cat <<EOF > src/simple/main.go
package main

import (
	"fmt"
	"github.com/gdamore/tcell"
)

func main() {
	fmt.Println(tcell.NewScreen)
}
EOF

cat <<EOF > src/hello/main.go
package main

import "fmt"

func main() {
	fmt.Println("hello")
}
EOF
GOPATH="$HOME/bugbb:$HOME/go" GO111MODULE=off makebb hello 
GOPATH="$HOME/bugbb:$HOME/go" GO111MODULE=off makebb hello simple
```

**Additional context**
This used to work it seems, so maybe this is related to go modules being broken.

hagna, 2022-01-06:
It will build a busybox with the simple in it if you checkout gobusybox and add simple to gobusybox/test/nested/cmd/simple and then run makebb test/nested/cmd/simple.

MDr164, 2022-02-21:
Can confirm. I'll be looking into it once I got time, as I need to build a package with this library as well now. A workaround would be to build the needed program separately and include it as an extra file. This however will increase the size of the initramfs quite a bit as the externally build program won't share the runtime with the busyboxed commands.

rminnich, 2022-06-08:
so as of today, I did this:
ran the script above
cd bug
cd src/
cd simple/
go mod init github.com/simple
go mod tidy
cd ../hello/
go mod init github.com/hello
go mod tidy
makebb hello
makebb hello simple

works.
rminnich-macbookpro2:src rminnich$ ./bb hello
hello
rminnich-macbookpro2:src rminnich$ ./bb simple
0x103001220

does this help at all?

I realize the modules usage is less than easy nowadays, but it did work ...

hugelgupf, 2024-02-25:
With #2923 our README will specifically outline what to do for modules and multi-modules.

closed 2024-02-27 by hugelgupf