u-root/u-root #2200: Cannot build u-root in Ubuntu 18.04 with go1.17.3

mcharleb, 2021-12-01:
**Describe the bug**
Running "go get github.com/u-root/u-root" using go1.17.3 causes u-root to fail

**To Reproduce**
```
# I have limited space under $HOME so I have to set GOPATH and GOCACHE
$ mkdir -p /local/mnt2/workspace2/go
$ mkdir -p /local/mnt2/workspace2/gocache
$ export GOPATH=/local/mnt2/workspace2/go
$ export GOCACHE=/local/mnt2/workspace2/gocache
$ Install go1.17.3 at /usr/local/go
$ export PATH=/usr/local/go/bin
$ git clone https://github.com/u-root/u-root
$ cd u-root
$ go get github.com/u-root/u-root
$ u-root
2021/12/01 14:23:50 Disabling CGO for u-root...
2021/12/01 14:23:50 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/local/mnt2/workspace2/go CGO_ENABLED=0
2021/12/01 14:23:50 "github.com/u-root/u-root/cmds/core/*" is neither package or path/glob: malformed import path "github.com/u-root/u-root/cmds/core/*": invalid char '*'
```

I can "fix" this by reverting to v1."go get github.com/u-root/u-root" and using go1.14.15
```
$ sudo rm -rf /local/mnt2/workspace2/gocache/* /local/mnt2/workspace2/go/*
$ git checkout v0.7.0
$ GOROOT=/opt/go1.14/go PATH=/opt/go1.14/go/bin:$PATH go get github.com/u-root/u-root
$ GOROOT=/opt/go1.14/go PATH=/opt/go1.14/go/bin:$PATH u-root
2021/12/01 14:39:05 Disabling CGO for u-root...
2021/12/01 14:39:05 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/opt/go1.14/go GOPATH=/local/mnt2/workspace2/go CGO_ENABLED=0
2021/12/01 14:39:05 WARNING: You are not using one of the recommended Go versions (have = go1.14.15, recommended = [go1.17]).
                        Some packages may not compile.
                        Go to https://golang.org/doc/install to find out how to install a newer version of Go,
                        or use https://godoc.org/golang.org/dl/go1.17 to install an additional version of Go.
2021/12/01 14:39:05 Skipping package "/local/mnt2/workspace2/go/src/github.com/u-root/u-root/cmds/core/bind": no buildable Go source files in /local/mnt2/workspace2/go/src/github.com/u-root/u-root/cmds/core/bind
2021/12/01 14:39:11 Successfully built "/tmp/initramfs.linux_amd64.cpio" (size 15895832).
$ git checkout master
$ go get github.com/u-root/u-root
$ u-root
2021/12/01 14:40:13 Disabling CGO for u-root...
2021/12/01 14:40:13 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/local/mnt2/workspace2/go CGO_ENABLED=0
2021/12/01 14:40:13 Skipping package "/local/mnt2/workspace2/go/src/github.com/u-root/u-root/cmds/core/bind": no buildable Go source files in /local/mnt2/workspace2/go/src/github.com/u-root/u-root/cmds/core/bind
2021/12/01 14:40:16 Successfully built "/tmp/initramfs.linux_amd64.cpio" (size 12524824).
```

**Expected behavior**
I would expect that it works with go1.17.3

mcharleb, 2021-12-02:
This looks like Issue https://github.com/u-root/u-root/issues/1859

Seems like the README.md doesn't make it clear that you need "export GO111MODULE=off", or that is it supposed to work without that, but does not.

Using go1.17.3, If I run
```
$ mkdir -p /local/mnt2/workspace2/go
$ mkdir -p /local/mnt2/workspace2/gocache
$ export GOPATH=/local/mnt2/workspace2/go
$ export GOCACHE=/local/mnt2/workspace2/gocache
# Install go1.17.3 at /usr/local/go
$ export PATH=/usr/local/go/bin
$ git clone https://github.com/u-root/u-root
$ cd u-root
$ export GO111MODULE=off
$ go get github.com/u-root/u-root
$ u-root
```
Then it works.

mcharleb, 2021-12-02:
After reading the very helpful https://maelvls.dev/go111module-everywhere/

I can run the following:
```
$ go install github.com/u-root/u-root@latest
$ u-root
2021/12/01 20:46:37 Disabling CGO for u-root...
2021/12/01 20:46:37 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/local/mnt2/workspace2/go CGO_ENABLED=0
2021/12/01 20:46:37 WARNING: You are not using one of the recommended Go versions (have = go1.17.3, recommended = [go1.13 go1.14]).
                        Some packages may not compile.
                        Go to https://golang.org/doc/install to find out how to install a newer version of Go,
                        or use https://godoc.org/golang.org/dl/go1.13 to install an additional version of Go.
2021/12/01 20:46:37 Filename is /tmp/initramfs.linux_amd64.cpio
2021/12/01 20:46:37 "github.com/u-root/u-root/cmds/core/*" is neither package or path/glob: malformed import path "github.com/u-root/u-root/cmds/core/*": invalid char '*'
```
Now it creates the archive but still complains about the wildcards.

I also noticed that the following fails:
```
 u-root minimal
2021/12/01 20:47:43 Disabling CGO for u-root...
2021/12/01 20:47:43 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/local/mnt2/workspace2/go CGO_ENABLED=0
2021/12/01 20:47:43 WARNING: You are not using one of the recommended Go versions (have = go1.17.3, recommended = [go1.13 go1.14]).
                        Some packages may not compile.
                        Go to https://golang.org/doc/install to find out how to install a newer version of Go,
                        or use https://godoc.org/golang.org/dl/go1.13 to install an additional version of Go.
2021/12/01 20:47:43 Filename is /tmp/initramfs.linux_amd64.cpio
2021/12/01 20:47:43 "github.com/u-root/u-root/cmds/core/installcommand" is neither package or path/glob: open /local/mnt2/workspace2/mcharleb/u-root/cmds/core/installcommand: no such file or directory
```
```

mcharleb, 2021-12-02:
The above did not seem to clone master. I ran the following, but still no luck running u-root:
```
$ export GO111MODULE=on
$ go install github.com/u-root/u-root@master
go: downloading github.com/u-root/u-root v0.8.1-0.20211202050031-390845ead28a
$ u-root                                                                                                      
2021/12/01 21:05:11 Disabling CGO for u-root...                                                              
2021/12/01 21:05:11 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/local/mnt2/workspace2/go CGO_ENABLED=0
2021/12/01 21:05:11 "github.com/u-root/u-root/cmds/core/*" is neither package or path/glob: malformed import path "github.com/u-root/u-root/cmds/core/*": invalid char '*'

rjoleary, 2021-12-02:
These issues should be fixed once https://github.com/u-root/u-root/pull/2168 is integrated.

orangecms, 2021-12-13:
Could you please recheck @mcharleb ? The branch to use is called `main` now. :-)

mcharleb, 2021-12-13:
On a fresh machine I installed go1.17.5, set PATH, cloned main branch and tried install and build instructions in README.md:
```
$ export PATH=/home/mcharleb/go/bin:/usr/local/go/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
$ git clone https://github.com/u-root/u-root
$ cd u-root
$ go get github.com/u-root/u-root
$ u-root 
2021/12/13 12:01:42 Disabling CGO for u-root...
2021/12/13 12:01:42 Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/home/mcharleb/go CGO_ENABLED=0
2021/12/13 12:01:42 NOTE: building with the new gobusybox; to get old behavior, use -build=bb
2021/12/13 12:01:42 "github.com/u-root/u-root/cmds/core/*" is neither package or path/glob: malformed import path "github.com/u-root/u-root/cmds/core/*": invalid char '*'
```
Same problem still exists.

orangecms, 2022-03-06:
Sorry for the long period of silence @mcharleb!

We more elaborate instructions now. I can reproduce the following steps:
https://github.com/u-root/u-root#usage

And I've opened #2342 to make it a bit more clear and in #2341 I've described how it can work using the new gobusybox build mode plus external commands as a workaround until the issue with that is fixed.

Does that help?

AfteratiadLily9, 2022-07-07:
I have same problem.  And it's still report the same error after set the path follow the readme . Go version 1.17.6 UBUNTU 20.04

rminnich, 2022-07-27:
I would say, since you cloned it anyway, try this:
cd to where you git cloned it
go build .
./u-root
can you see if that works.
also as an experiment unset GOPATH

sorry this has been so painful, modules has been a total shitshow for a few years now and we're just about struggling through it but still hitting some issues.

hugelgupf, 2024-02-25:
The correct invocation with Go modules is `go install github.com/u-root/u-root@latest`. This is fixed by #2923.

closed 2024-02-27 by hugelgupf