u-root/u-root #2257: CI: amd64 integration image lacking /etc/group file

ChriMarMe, 2022-01-05:
**Describe the bug**
AMD64 integration image missing /etc/group file. 

**To Reproduce**
Write a function which tries to open /etc/group file.

**Expected behavior**
File existence

**Additional context**
Code which requires package std os/user can't rely on proper integration tests if the matching of group id to group name does not work properly.

rjoleary, 2022-01-05:
I think you would need to create a temporary group file with some fake data and add the file to the following struct: https://github.com/9elements/u-root/blob/main/integration/gotests/gotest_test.go#L95

See how it was done in webboot: https://github.com/u-root/webboot/blob/main/integration/basic_test.go#L56

ChriMarMe, 2022-01-05:
Would this create side effects of the other integration tests?

ChriMarMe, 2022-01-05:
Ok, fix worked. Will push that soon

closed 2022-01-05 by ChriMarMe