u-root/u-root #2300: bzimage package only parses gz compression, and must support more options

rminnich, 2022-02-02:
**Describe the bug**
the bzimage package can't unpack kernels compress with xz. This is because when I wrote it I was not sure how to find
the compressed part and used a simple minded search. 

**To Reproduce**
create a bzImage but enable XZ compression in the kernel (yep, it's still called a bzimage)
use the kexec command with debug and note you see this:
2022/02/02 21:13:15 Failed to unpack compressed kernelCode: flate: corrupt input before offset 1

oops.

**Expected behavior**
it unpacks correctly. 

**Additional context**
we'll need to better unpack bzimages; see coreboot's unpacker for help maybe?

10000TB, 2022-02-22:
To clarify, this does not work even we build a  `/usr/bin/xzcat`  into initramfs ?

abrender, 2022-08-11:
FYI I've been playing around with [bzimage.go](https://github.com/u-root/u-root/blob/main/pkg/boot/bzimage/bzimage.go) and I've found a few things that I'll note down here before trying to figure out any fixes:

Reference: https://www.kernel.org/doc/Documentation/x86/boot.txt

There is a 4 byte CRC-32 checksum added to the end of the file.    I believe that this explains the ["String of unknown meaning" - aka `initRAMFStag`](https://github.com/u-root/u-root/blob/main/pkg/boot/bzimage/bzimage.go#L56)  and the [commented out code](https://github.com/u-root/u-root/blob/main/pkg/boot/bzimage/bzimage.go#L254) that doesn't check the stderr (or the return code) of the decompression - the decompression always returns an error because there are 4 extra bytes (the CRC) sent to be decompressed. 

Changing ([here](https://github.com/u-root/u-root/blob/main/pkg/boot/bzimage/bzimage.go#L140)) `b.compressed = make([]byte, b.Header.PayloadSize)` to `b.compressed = make([]byte, b.Header.PayloadSize - 4)` makes things much happier with regards to error checking and decompressing.

If the code is cleaned up to properly handle the 4-byte CRC then it's possible that we can change the exec.Command usages with go packages as suggested by @10000TB in the above comment  (possibly creating a type `type decompressor (io.Reader, io.WRiter) error` and using that instead of *exec.Cmd in the `magic` struct.); that'll save the overhead of executing a command and the complexity of having to handle stdin, stdout, stderr, return codes, etc.

abrender, 2022-08-17:
Okay, I lied in the comment above.   There is indeed a 4 byte CRC32 checksum at the end of the file, but that's not what's causing the decompression to fail.

The decompression fails because the compressed payload is actually [compressed vmlinux.bin.all + u32 size of vmlinux.bin.all](https://github.com/torvalds/linux/blob/master/arch/x86/boot/compressed/Makefile#L18) which can be [verified in arch/x86/boot/compressed/mkpiggy.c](https://github.com/torvalds/linux/blob/master/arch/x86/boot/compressed/mkpiggy.c#L40-L47).   So we should strip off the last 4 bytes of the payload, decompress, then compare the size.

I'll send a set of fixes to clean up the test code, add CRC32 support and then fix this decompression bug.

PR #2481 (2022-08-25): bzimage: Fix broken decompression.
The (poorly documented) compressed kernel payload is actually
the compressed payload + 4 bytes containing the uncompressed size ([1]
[2]). The code in bzimage.go assumed that these 4 trailing bytes were
part of the compressed payload, which lead to (1) decompression always
returning an error (see #2300) (2) commenting out of code to check
stderr of the decompressor (3) inclusion of the "String of unknown
meaning" aka initRAMFStag in the code.

This code (1) adds in the proper parsing of the trailing 4 bytes (2)
rewrites the code used when swapping the compres [... cut]
commit 4a3d3e86 (2022-08-25): bzimage: Fix broken decompression.

The (poorly documented) compressed kernel payload is actually
the compressed payload + 4 bytes containing the uncompressed size ([1]
[2]). The code in bzimage.go assumed that these 4 trailing bytes were
part of the compressed payload, which lead to (1) decompression always
returning an error (see #2300) (2) commenting out of code to check
stderr of the decompressor (3) inclusion of the "String of unknown
meaning" aka initRAMFStag in the code.

This code (1) adds in the proper parsing of the trailing 4 bytes (2)
rewrites the code used when swapping the compre [... cut]
closed 2022-08-25 by probot-auto-merge[bot]