u-root/u-root #2326: Saving a GPT partition table is broken

DavidVentura, 2022-02-20:
**Describe the bug**
Opening a valid disk image and saving it back will break the partition table.

gdisk output on the new file:
```
Recovery/transformation command (? for help): v

Problem: The secondary header's self-pointer indicates that it doesn't reside
at the end of the disk. If you've added a disk to a RAID array, use the 'e'
option on the experts' menu to adjust the secondary header's and partition
table's locations.

Problem: Disk is too small to hold all the data!
(Disk size is 131071 sectors, needs to be 131072 sectors.)
The 'e' option on the experts' menu may fix this problem.

Identified 2 problems!

```
**To Reproduce**

```go
package main

import (
        "io"
        "os"

        "github.com/u-root/u-root/pkg/mount/gpt"
)

func main() {
        f, _ := os.Open("hdd.raw")
        r := io.ReaderAt(f)
        t, _ := gpt.New(r)
        new_f, _ := os.OpenFile("hdd2.raw", os.O_WRONLY|os.O_CREATE, 0755)
        gpt.Write(new_f, t)
}

```

To create `hdd.raw`:

```
$ dd if=/dev/zero bs=1M count=64 of=hdd.raw
$ gdisk hdd.raw
Command (? for help): n
Partition number (1-128, default 1): 
First sector (34-131038, default = 2048) or {+-}size{KMGTP}: 
Last sector (2048-131038, default = 131038) or {+-}size{KMGTP}: 4096
Current type is 8300 (Linux filesystem)
Hex code or GUID (L to show codes, Enter = 8300): 
Changed type of partition to 'Linux filesystem'

Command (? for help): p
Disk hdd.raw: 131072 sectors, 64.0 MiB
Sector size (logical): 512 bytes
Disk identifier (GUID): F8867B9A-94D9-46B9-AC1C-E088F5E39B0E
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 131038
Partitions will be aligned on 2048-sector boundaries
Total free space is 128956 sectors (63.0 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048            4096   1024.5 KiB  8300  Linux filesystem
```

Size of files differs by 420 bytes ??
```
67108444 feb 20 13:16 hdd2.raw
67108864 feb 20 13:16 hdd.raw
```

**Expected behavior**
I expect writing an unmodified table to generate identical files

**Additional context**
Add any other context about the problem here.

abrender, 2022-08-02:
"Size of files differs by 420 bytes ??"

https://en.wikipedia.org/wiki/GUID_Partition_Table  - the GPT header is 512 bytes (sector size), 92 of which are reserved, and then there's 420 bytes of "Reserved; must be zeroes for the rest of the block".

gpt.go only writes the 92 allocated bytes of the GPT header, not the full 512.   The 420 reserved zero bytes are not written.

Your reproduction code creates a new hdd2.raw file - but if you pre-create it with the same size as hdd.raw (and therefore zeroing the remaining 420 bytes), you'll see that it writes the exact same contents:

```
$ ls -ld *.raw
-rw-r----- 1 abrender primarygroup 67108864 Aug  1 11:41 hdd.raw

# Create a new hdd2.raw file of the same size.
$ dd if=/dev/zero bs=1M count=64 of=hdd2.raw
64+0 records in
64+0 records out
67108864 bytes (67 MB, 64 MiB) copied, 0.046969 s, 1.4 GB/s

# The files have the same size, but different checksum:
$ ls -ld *.raw; md5sum *.raw
-rw-r----- 1 abrender primarygroup 67108864 Aug  2 07:09 hdd2.raw
-rw-r----- 1 abrender primarygroup 67108864 Aug  1 11:41 hdd.raw
7f614da9329cd3aebf59b91aadc30bf0  hdd2.raw
783b81eaa2bd17ad14eab490a8cbc6fe  hdd.raw

$ go run gpt.go

# The files have the same size and checksum.
$ ls -ld *.raw; md5sum *.raw
-rw-r----- 1 abrender primarygroup 67108864 Aug  2 07:10 hdd2.raw
-rw-r----- 1 abrender primarygroup 67108864 Aug  1 11:41 hdd.raw
783b81eaa2bd17ad14eab490a8cbc6fe  hdd2.raw
783b81eaa2bd17ad14eab490a8cbc6fe  hdd.raw
```

The code change for writing the full sector, instead of just the 512 bytes, is pretty small: https://gist.github.com/abrender/767e4a02c2524f3de9d69fadda7a48a8 . I'll check into adding some tests and then submitting a pull request.

10000TB, 2022-08-02:
Following the repro code, I seem to see a panic 

```
xuehaohu@worldpeace:~/go/src/github.com/u-root/u-root$ go run repro.go
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x50 pc=0x4adc9e]

goroutine 1 [running]:
github.com/u-root/u-root/pkg/mount/gpt.writeGPT({0x4f1f48, 0xc000012040}, 0x0)
	/usr/local/google/home/xuehaohu/go/src/github.com/u-root/u-root/pkg/mount/gpt/gpt.go:287 +0x3e
github.com/u-root/u-root/pkg/mount/gpt.Write({0x4f1f48, 0xc000012040}, 0xc000010030)
	/usr/local/google/home/xuehaohu/go/src/github.com/u-root/u-root/pkg/mount/gpt/gpt.go:274 +0x65
main.main()
	/usr/local/google/home/xuehaohu/go/src/github.com/u-root/u-root/repro.go:15 +0x6c
exit status 2
```

10000TB, 2022-08-02:
Under a sector of 512bytes, I can make sense of the 420 bytes difference, as explained

PR #2461 (2022-08-19): GPT: Read/Write GPT header in whole-sector increment.
The current code reads/writes GPT headers in 92-byte `HeaderSize`
increments. Changing the code to read/write GPT headers in sector
increments is a tradeoff with the following pros/cons:

Reading the whole sector:
* Pro: Removes a use of hard coded 0x5c value of `HeaderSize`.
       Makes it simpler to support different header sizes in the future.
* Con: Requires reading an additional 820 bytes (420 for primary, 420
       for secondary).

Writing the whole sector:
* Pro: Enforces the expectation/requirement that the unused 420 reserved
    [... cut]
closed 2022-08-19 by probot-auto-merge[bot]