u-root/u-root #2330: "GO111MODULE=off go get github.com/u-root/u-root" not working

StuartRothrock, 2022-02-26:
**Describe the bug**
Running "GO111MODULE=off go get github.com/u-root/u-root" as suggested in the README has no results. No output and no dirs/files produced.

**To Reproduce**
execute GO111MODULE=off go get github.com/u-root/u-root

**Expected behavior**
It should compile properly.

**Additional context**
Installed go version go1.17.7 linux/amd64

StuartRothrock, 2022-02-26:
I started with a fresh $GOHOME and now I get the following.
```
# github.com/u-root/u-root/pkg/lockfile
../.gohome/src/github.com/u-root/u-root/pkg/lockfile/lockfile.go:82:18: undefined: os.CreateTemp
../.gohome/src/github.com/u-root/u-root/pkg/lockfile/lockfile.go:142:20: undefined: os.ReadFile
# github.com/u-root/u-root/pkg/uio
../.gohome/src/github.com/u-root/u-root/pkg/uio/reader.go:29:9: undefined: io.ReadAll
```

orangecms, 2022-03-06:
Thank you for the issue! Can you elaborate a little? There should be `/home/administrator/.gohome/bin/u-root` after the command you ran, was that not created?

The README currently only says "make sure your `GOPATH` is set up correctly, but doesn't elaborate further, sorry, gotta fix that! =)

StuartRothrock, 2022-03-06:
Thanks for the reply. There are no files in the /home/administrator/.gohome
folder.  I didn't get any hits related to it with a google search. What am
I doing wrong?

orangecms, 2022-03-06:
Well, what _do_ you get? What is the output in the terminal from the go get command? Just nothing? That'd mean success. And did you try `GOPATH` as the README suggests? `GOHOME` is something different, I think.

Do you have anything in `~/go/bin`?

StuartRothrock, 2022-03-06:
I am embarrassed for the "dumb" user problem. I had a typo for GOPATH in my run script. Now I am getting pretty far. Now I get this far but I'll try to figure it out. 
```
Build environment: GOARCH=amd64 GOOS=linux GOROOT=/usr/local/go GOPATH=/home/administrator/go CGO_ENABLED=0
Skipping package "/home/administrator/go/src/github.com/u-root/u-root/cmds/core/bind": no buildable Go source files in /home/administrator/go/src/github.com/u-root/u-root/cmds/core/bind
```

orangecms, 2022-03-06:
Awesome, you succeeded building a u-root cpio image! \o/ :partying_face: 

Now you can test and run it e.g. via QEMU :-)

https://github.com/u-root/u-root#testing-in-qemu

closed 2022-03-06 by StuartRothrock