u-root/u-root #2356: GRUB environemnt file parsing does not handle escapes correctly

zyga, 2022-03-15:
**Describe the bug**

I wrote a GRUB env handling library and I'm looking at the available modules to report issues that they handle escape logic incorrectly. GRUB values may contain newlines or escaped `/` characters. Sadly this is often neglected resulting in a possibility to interpret the same file differently in userspace with 3rd party tools and differently in GRUB itself.

**To Reproduce**
Steps to reproduce the behavior:

Manipulate any file with a value containing a newline or `/` character.

**Expected behavior**

The value should be handled correctly.

**Additional context**

Consider using 
https://pkg.go.dev/gitlab.com/zygoon/go-grub@v0.1.0 which handles this (and possible more) edge cases correctly.

hugelgupf, 2022-03-16:
Your library seems to force users to use file system files. We need an API that uses io.Reader and io.Writers.

(GRUB has functionality to support netboot, in which case files can be non-local-file-system files not opened with os.Open.)

zyga, 2022-03-16:
The `Env` type in my library can be combined by any reader/writer, it is just a `[1024]byte` type.

Can you explain how a io.Reader and io.Writer integration you are looking for would look like?

zyga, 2022-03-18:
I'll reach out again saying I'm open to extending the API if that would help you. Since `Env` can be treated as `[]byte` and written to any `io.Writer` or read from any `io.Reader`, I didn't add any special methods but if there is something I'm missing please tell me about it.

rminnich, 2022-03-29:
func LoadEnv(path string) (*Env, error) {
could this take an io.Reader, and maybe
LoadEnvFile(path string)
that sort of change, wdyt?

zyga, 2022-03-31:
Let me propose something and ping back with a MR/diff, would you be able to review it?