u-root/u-root #2365: Echo fails to write into sysfs

yurinnick, 2022-03-28:
**Describe the bug**
`echo` command can't write data into sysfs, however `tee` works just fine.
I suspect that's because `echo` uses `Fprinf()` into file, while `tee` uses `io.Copy()`.

Per sysfs documentation:
```
On write(2), sysfs expects the entire buffer to be passed during the first write. 
Sysfs then passes the entire buffer to the store() method.
A terminating null is added after the data on stores.
This makes functions like sysfs_streq() safe to use.
```

**To Reproduce**
```
$ echo powersave > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
Exception: write /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor: invalid argument
[tty 1], line 1: echo powersave > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor

$ echo powersave | tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
powersave
```

**Expected behavior**
Coreutils echo has no problem writing into sysfs files.

**Additional context**
- My working implementation is [here](https://github.com/yurinnick/pinephone-uroot/blob/main/jumpdrive/jumpdrive.go#L108=)
- [Sysfs: Reading/Writing Attribute Data](https://www.kernel.org/doc/html/latest/filesystems/sysfs.html#reading-writing-attribute-data)

rminnich, 2022-03-29:
I just tried to reproduce this. I can't.

I'm here: commit f5ea5f932ab92708f024833174268c57af737cda
go build .
root@a300:/home/rminnich/go/src/github.com/u-root/u-root/cmds/core/echo# ./echo powersave > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
root@a300:/home/rminnich/go/src/github.com/u-root/u-root/cmds/core/echo#

When I run it with strace, I see this:
write(1, "powersave\n", 10)             = 10

Can you run under strace, and let me know what you see?

yurinnick, 2022-03-30:
I guess I missed an important detail - I am running it on the Pinephone. Since it's `arm64` there's no `strace` available there.
Thankfully the issue can be reproduced in QEMU:

```
$ qemu-system-aarch64 -M virt -kernel ./output/Image -initrd ./output/initramfs-uroot.cpio -append "console=ttyAMA0,115200" -cpu cortex-a57 -M virt -m 1024 -nographic
...
/# cat /sys/block/loop0/queue/scheduler
[none] bfq

/# echo bfq > /sys/block/loop0/queue/scheduler
Exception: write /sys/block/loop0/queue/scheduler: invalid argument
[tty 11], line 1: echo bfq > /sys/block/loop0/queue/scheduler

/# echo bfq | tee /sys/block/loop0/queue/scheduler
bfq

/# cat /sys/block/loop0/queue/scheduler
[bfq] none
```

yurinnick, 2022-03-30:
Is it possible to run `strace`over `cpud`?